P0cL4bs/WiFi-Pumpkin-deprecated #120: Kubuntu: No Module Named QtWebKit

invious, 2016-11-26:
```
kubuntu@kubuntu:~/WiFi-Pumpkin$  sudo apt-get install python-qt4 
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Suggested packages:
  python-qt4-dbg
The following NEW packages will be installed:
  python-qt4
0 upgraded, 1 newly installed, 0 to remove and 99 not upgraded.
Need to get 0 B/2,223 kB of archives.
After this operation, 14.2 MB of additional disk space will be used.
Selecting previously unselected package python-qt4.
(Reading database ... 173484 files and directories currently installed.)
Preparing to unpack .../python-qt4_4.11.4+dfsg-2_amd64.deb ...
Unpacking python-qt4 (4.11.4+dfsg-2) ...
Setting up python-qt4 (4.11.4+dfsg-2) ...
kubuntu@kubuntu:~/WiFi-Pumpkin$ sudo wifi-pumpkin
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 50, in <module>
    from core.main import Initialize
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 52, in <module>
    from core.helpers.report import frm_ReportLogger
  File "/usr/share/WiFi-Pumpkin/core/helpers/report.py", line 2, in <module>
    from PyQt4.QtWebKit import QWebView
ImportError: No module named QtWebKit
```

mh4x0f, 2016-11-26:
It looks like the PyQt4.QtWebKit package got removed in Ubuntu 16.10, probably because both PyQt4 and QtWebKit are long unmaintained. 
from  [stackoverflow](http://askubuntu.com/a/840946)

mh4x0f, 2016-11-26:
I'll add fix coming soon

closed 2016-11-27 by mh4x0f