P0cL4bs/WiFi-Pumpkin-deprecated #121: TypeError: unable to convert a QVariant of type 0 to a QMetaType of type 10

CptKirk, 2016-11-26:
Everytime I want to start an AP in the Terminal pops-up an error message:

```
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1079, in Start_PumpAP
    if not type(self.SoftDependencies()) is bool: return
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1067, in SoftDependencies
    self.hostapd_path = self.FSettings.Settings.get_setting('accesspoint','hostapd_path')
  File "/usr/share/WiFi-Pumpkin/core/utility/collection.py", line 33, in get_setting
    value = self.psettings.value(key,type=format)
TypeError: unable to convert a QVariant of type 0 to a QMetaType of type 10

```

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): RTL8723BE 
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): no
* Operating System and version: Ubuntu 14.04 LTS

mh4x0f, 2016-11-27:
check whitch version you are using from PyQt. you can see this in Help->About WiFi-Pumpkin-> Version.

CptKirk, 2016-11-27:
QtVersion: 4.8.6

closed 2016-11-27 by mh4x0f