P0cL4bs/WiFi-Pumpkin-deprecated #124: web cloner becomes non-responsive

denux, 2016-12-03:
I created a beef hook and in DNS spoof module, I then select a target.
i phishing manager I paste my hook and want to open facebook login with hook when targets opens example.com as domain.
but the phishing manager stops responding, I have tried multiple combinations it works when i use index.html but becomes non-responsive on choosing website clone and entering "facebook.com/login" or other links.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

mh4x0f, 2016-12-04:
look [bug reppoting](https://github.com/P0cL4bs/WiFi-Pumpkin/wiki/Bug-reporting)

closed 2016-12-04 by mh4x0f

mh4x0f, 2016-12-04:
@denux I don't find any problem with this module, i try in my lab with different OS and all working fine.The phishing module  is crashing  because you finishing the connection with clients connected  the HTTP server can't finish the process and return a error in terminal.

jeretc, 2016-12-14:
i made a fake-portal to go with wifi-pumpkin. have tested it and works well. i used these modules in wifi-pumpkin, DNS Spoof and Phising Manager with the Rogue AP. if you ever want to try it out ..

https://github.com/jeretc/fake-portal