P0cL4bs/WiFi-Pumpkin-deprecated #125: Cant create ap and stoppıng all server

bayp1p0, 2016-12-03:
http://i.hizliresim.com/vXLppv.jpg
I get this error with use 2 adapter. 
My system parrot 3.2
How can ı solve this error

bayp1p0, 2016-12-03:
And some times ı can create ap but in this time wifi-pumpkın cant see connecting clıent.

denux, 2016-12-03:
-go to etc/networkmanager/networkmanager.config
-after key file add this 
[keyfile]
unmanaged-devices=wlan1
-save n exit
-in terminal type "service network-manager restart" (without quote)
-connect the internet with wlan0
-run wifi-pumpkin n choose wlan1 for AP.

hope it helped.
thank you.

mh4x0f, 2016-12-04:
you can fix  the network-manager config, see in this comment issue [solution](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/89#issuecomment-254022958)

closed 2016-12-04 by mh4x0f