P0cL4bs/WiFi-Pumpkin-deprecated #127: failed to initiate accesss point.

jeretc, 2016-12-04:
## What's the problem (or question)?

*have attached the image file. in my hostapd.conf is this :

interface=wlan1
ssid=PumpAP
channel=11
driver=nl80211
*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):tplink 722n

* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): no
* Operating System and version: parrotsec 3
[image]

mh4x0f, 2016-12-04:
see in this comment issue [solution](https://github.com/P0cL4bs/WiFi-Pumpkin/issues/89#issuecomment-254022958)

closed 2016-12-04 by mh4x0f

jeretc, 2016-12-04:
hi using wicd solved the issue.

jeretc, 2016-12-07:
Hi is there a manual for wifi-pumpkin.

Thanks.