P0cL4bs/WiFi-Pumpkin-deprecated #130: Responder - Error starting TCP server on port 443

userdsp, 2016-12-06:
## What's the problem (or question)?

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):wn722n and AWUS036H
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): yes kali linux 2016.2 64bit
* Operating System and version: Linux kali 4.8.0-kali1-amd64 #1 SMP Debian 4.8.5-1kali1 (2016-11-04) x86_64 GNU/Linux
root@kali:~/handshakes# responder -i mon0
                                         __
  .----.-----.-----.-----.-----.-----.--|  |.-----.----.
  |   _|  -__|__ --|  _  |  _  |     |  _  ||  -__|   _|
  |__| |_____|_____|   __|_____|__|__|_____||_____|__|
                   |__|

           NBT-NS, LLMNR & MDNS Responder 2.3.3.1

  Author: Laurent Gaffie (laurent.gaffie@gmail.com)
  To kill this script hit CRTL-C

Error: -I <if> mandatory option is missing
root@kali:~/handshakes# responder -I mon0
                                         __
  .----.-----.-----.-----.-----.-----.--|  |.-----.----.
  |   _|  -__|__ --|  _  |  _  |     |  _  ||  -__|   _|
  |__| |_____|_____|   __|_____|__|__|_____||_____|__|
                   |__|

           NBT-NS, LLMNR & MDNS Responder 2.3.3.1

  Author: Laurent Gaffie (laurent.gaffie@gmail.com)
  To kill this script hit CRTL-C

[+] Poisoners:
    LLMNR                      [ON]
    NBT-NS                     [ON]
    DNS/MDNS                   [ON]

[+] Servers:
    HTTP server                [ON]
    HTTPS server               [ON]
    WPAD proxy                 [OFF]
    Auth proxy                 [OFF]
    SMB server                 [ON]
    Kerberos server            [ON]
    SQL server                 [ON]
    FTP server                 [ON]
    IMAP server                [ON]
    POP3 server                [ON]
    SMTP server                [ON]
    DNS server                 [ON]
    LDAP server                [ON]

[+] HTTP Options:
    Always serving EXE         [OFF]
    Serving EXE                [OFF]
    Serving HTML               [OFF]
    Upstream Proxy             [OFF]

[+] Poisoning Options:
    Analyze Mode               [OFF]
    Force WPAD auth            [OFF]
    Force Basic Auth           [OFF]
    Force LM downgrade         [OFF]
    Fingerprint hosts          [OFF]

[+] Generic Options:
    Responder NIC              [mon0]
    Responder IP               [192.168.1.106]
[... 105 lines cut ...]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,454 : [+] HTTP Options:
SessionID[NDA4ODE=] 2016-12-05 11:55:28,456 :     Always serving EXE         [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,458 :     Serving EXE                [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,460 :     Serving HTML               [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,463 :     Upstream Proxy             [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,465 : 
SessionID[NDA4ODE=] 2016-12-05 11:55:28,467 : [+] Poisoning Options:
SessionID[NDA4ODE=] 2016-12-05 11:55:28,471 :     Analyze Mode               [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,473 :     Force WPAD auth            [ON]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,476 :     Force Basic Auth           [ON]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,478 :     Force LM downgrade         [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,480 :     Fingerprint hosts          [OFF]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,483 : 
SessionID[NDA4ODE=] 2016-12-05 11:55:28,485 : [+] Generic Options:
SessionID[NDA4ODE=] 2016-12-05 11:55:28,487 :     Responder NIC              [wlan0mon]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,490 :     Responder IP               [10.0.0.1]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,492 :     Challenge set              [1122334455667788]
SessionID[NDA4ODE=] 2016-12-05 11:55:28,495 :     Don't Respond To Names     ['ISATAP']
SessionID[NDA4ODE=] 2016-12-05 11:55:28,497 : 
SessionID[NDA4ODE=] 2016-12-05 11:55:28,502 : 
SessionID[NDA4ODE=] 2016-12-05 11:55:28,505 : 
SessionID[NDA4ODE=] 2016-12-05 11:55:28,507 : [!] Error starting TCP server on port 80, check permissions or other servers running.
SessionID[NDA4ODE=] 2016-12-05 11:55:28,510 : [!] Error starting SSL server on port 443, check permissions or other servers running.
SessionID[NDA4ODE=] 2016-12-05 11:55:28,513 : [+] Listening for events...
SessionID[NDA4ODE=] 2016-12-05 11:55:28,516 :

closed 2016-12-06 by mh4x0f

mh4x0f, 2016-12-06:
[image]

userdsp, 2016-12-11:
The problem is [!] Error starting SSL server on port 443, check, but port 443 is not used by any other service!