P0cL4bs/WiFi-Pumpkin-deprecated #137: hostapd fails to start

TimSchumi, 2016-12-11:
## What's the problem (or question)?

When I try to start the AP, it fails with the following error:

Failed to initiate Access Point, check output process hostapd.

Output:
Configuration file: /usr/share/WiFi-Pumpkin/settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabed_11b_rates=0
nl80211: driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABED
wlan0: AP-DISABLED
hostapd_free_hapd_data: Interface wlan0 wasn't started

Running "airmon-ng check kill" does solve this problem (Source: http://askubuntu.com/questions/472794/hostapd-error-nl80211-could-not-configure-driver-mode)
Is it possible to run this (or any other solution) automatically, when this error occurs?

"airmon-ng check kill" just terminated wpa_supplicant and dhclient and the AP works again, so it would be possible to kill off one of then (or both of them) so that airmon is not needed

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): RTL8821AE
* Version used tool: 0.8.3
* Virtual Machine (yes or no and which): no
* Operating System and version: Kali Rolling (with latest updates)

mh4x0f, 2016-12-11:
Do you using local connection or wireless ? because the tool using this command bellow for kill wpa_supplicant and dhclient, if you have ethernet connection.
``` sh
sudo nmcli nm wifi off
sudo rfkill unblock wlan
```
I don't want to use airmon-ng. i recommend exclude card from network-manager , look this solution #127

mh4x0f, 2016-12-11:
I'll add function for find this process network and kill it

closed 2016-12-12 by TimSchumi