P0cL4bs/WiFi-Pumpkin-deprecated #138: This doesn't work in my linux mint rosa (ubuntu 14.04)

merlinnusr, 2016-12-12:
I use this command 
>sudo  wifi-pumpkin

this message is displayed 

>PyQt4 app looks different when running with root, because of that the GUI not work 100%, some features not work. I don't find anything code or settings for fix this bug (if you have any solution for this send me feedback :D).
if you want keep the normal user style, run app with "sudo".

merlinnusr, 2016-12-12:
>CPU~Dual core Intel Celeron N2830 (-MCP-) speed/max~1199/2415 MHz Kernel~3.19.0-32-generic x86_64 Up~32 min Mem~1095.5/1884.5MB HDD~500.1GB(13.6% used) Procs~170 Client~Shell inxi~2.2.28

merlinnusr, 2016-12-12:
CPU~Dual core Intel Celeron N2830 (-MCP-) speed/max~1199/2415 MHz Kernel~3.19.0-32-generic x86_64 Up~32 min Mem~1095.5/1884.5MB HDD~500.1GB(13.6% used) Procs~170 Client~Shell inxi~2.2.28

merlinnusr, 2016-12-12:
CPU Dual core Intel Celeron N2830 (-MCP-) speed/max 1199/2415 MHz Kernel 3.19.0-32-generic x86_64 Up 32 min Mem 1095.5/1884.5MB HDD 500.1GB(13.6% used) Procs 170 Client Shell inxi 2.2.28

TimSchumi, 2016-12-12:
What happens if you start it by using the Desktop Icon? Also, wifi-pumpkin should show the sudo prompt automatically, so there is no need to do "sudo wifi-pumpkin". Just "wifi-pumpkin" should be enough.

merlinnusr, 2016-12-13:
@TimSchumi This request me a password but doesn't happen after

merlinnusr, 2016-12-13:
@TimSchumi This request me a password but nothing happens

TimSchumi, 2016-12-13:
And what happens if you put in the password of your account? If this doesn't work, there is something wrong with your system

mh4x0f, 2016-12-13:
@merlinnusr go to directory the WiF-Pumpkin (**/usr/share/WiFi-Pumpkin/**) and remove this [line](https://github.com/P0cL4bs/WiFi-Pumpkin/blob/master/wifi-pumpkin.py#L48) the file wifi-pumpkin.py

merlinnusr, 2016-12-13:
I know what is the problem I have three versions of pip I modify the script to force pip2 

>	pip install -r requirements.txt

>	pip2 install -r requirements.txt

merlinnusr, 2016-12-13:
https://github.com/P0cL4bs/WiFi-Pumpkin/blob/master/installer.sh

closed 2016-12-13 by merlinnusr