P0cL4bs/WiFi-Pumpkin-deprecated #142: wifi-pumpkin port to nethunter?

jeretc, 2016-12-16:
## What's the problem (or question)?

*Hi this is not an issue. Will there be a nethunter version ? would be great on nethunter. thanks.*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

TimSchumi, 2016-12-16:
If python and all the other dependencies are available for NetHunter, the program will most likely work too. The only thing that needs to be checked, is if a external WiFi Card is needed

jeretc, 2016-12-16:
Yes it requires a compatible external wifi card. I have that too.  Would
like yo test it out.

Thanks

closed 2016-12-16 by mh4x0f

mh4x0f, 2016-12-16:
No.