P0cL4bs/WiFi-Pumpkin-deprecated #146: Internet Proxy on the system running WiFi-Pumpkin

TimSchumi, 2016-12-22:
Is there a possibility to include a "Proxy Settings" Part into the Program, for the case a Proxy is needed to connect to a network to get Internet access on the Host System?

There should be the following Settings:
Proxy Host
Proxy Port
Proxy Username (optional)
Proxy Password (optional)

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

mh4x0f, 2016-12-22:
No, this is not interactive man-in-the-middle attack and it's not the purpose of the project. the model proxy for tool is transparent Proxy, because simulate real access wifi networks.

TimSchumi, 2016-12-23:
No, I meant if the host system on which WiFi-Pumpkin is running needs to connect through a Proxy

mh4x0f, 2016-12-23:
I understand now, But the system needs to connect through a proxy. I think that it must be done manually by the user. even using iptables but only a transparent proxy would be possible. I don't think it's viable.

closed 2016-12-23 by mh4x0f