P0cL4bs/WiFi-Pumpkin-deprecated #149: Not found file NetworkManager.conf in folder /etc/NetworkManager/

craftsmany, 2016-12-24:
## What's the problem (or question)?

When I start wifi pumpkin (the fake ap) it says "Not found file NetworkManager.conf in folder /etc/NetworkManager/" but the file is there.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
Yes it supports AP.
* Version used tool:
v0.8.4 
* Virtual Machine (yes or no and which): 
No
* Operating System and version: 
Linux Mint 18
(uname -a: Linux lunix 4.4.0-45-generic #66-Ubuntu SMP Wed Oct 19 14:12:37 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux)

mh4x0f, 2016-12-25:
send me output the command bellow:
``` 
cat /etc/NetworkManager/NetworkManager.conf
```

craftsmany, 2016-12-25:
[main]
plugins=ifupdown,keyfile,ofono
dns=dnsmasq

[ifupdown]
managed=false

mh4x0f, 2016-12-25:
it very strange, this check file with python code
``` sh
python -c "import os; print os.path.exists('/etc/NetworkManager/NetworkManager.conf')"
```
check if this code return True or False. if return False try change the code "os.path.exists" for  os.path.isfile.

craftsmany, 2016-12-26:
it returned true.

mh4x0f, 2016-12-26:
i did somes improvements in code, but maybe the problem is permission the file.

closed 2016-12-28 by mh4x0f