P0cL4bs/WiFi-Pumpkin-deprecated #162: cann't start port 443

cbogithub, 2017-01-03:
wifi pumpkin cann't start port 443 
https://postimg.org/image/bznpyb543/

#### Please tell us details about your environment.

    Card wireless adapters name (please check if support AP/mode): TL-WN821N
    Version used tool: 0.8.4
    Virtual Machine (yes or no and which): VMWare
    Operating System and version: Kali linux
[](url)

mh4x0f, 2017-01-03:
I fix only the bug when started responder plugin, the error "cannot start port 443 " is a problem your system.

closed 2017-01-03 by mh4x0f

cbogithub, 2017-01-04:
i have just updated it but still got erorr :  

```
root@kali:/home/WiFi-Pumpkin# wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [MzQ3MTE=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 4096 (hostapd)]
[New Thread 4098 (Net-Creds)]
[New Thread 4100 (Responder)]
[New Thread 4126 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 305, in run
    self.connector = reactor.listenTCP(int(listenPort), strippingFactory)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 478, in listenTCP
    p.startListening()
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 983, in startListening
    raise CannotListenError(self.interface, self.port, le)
twisted.internet.error.CannotListenError: Couldn't listen on any:10000: [Errno 98] Address already in use.
-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[e8:de:27:05:a3:82] CH 11
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[Net-Creds] successfully stopped.
Thread::[Responder] successfully stopped.
Thread::[Dns2Proxy] successfully stopped.
Thread::[sslstrip2] successfully stopped.

[*] Loading debugging mode
[*] Current Session::ID [NjE0NTA=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 4370 (hostapd)]
[New Thread 4373 (Net-Creds)]
[New Thread 4375 (Responder)]
sslstrip 0.9 + sergio-proxy v0.2.1 online
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 432, in run
    self.connectorSP = reactor.listenTCP(int(listenPort), strippingFactory)
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 478, in listenTCP
    p.startListening()
  File "/usr/local/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 983, in startListening
    raise CannotListenError(self.interface, self.port, le)
twisted.internet.error.CannotListenError: Couldn't listen on any:10000: [Errno 98] Address already in use.
-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[e8:de:27:05:a3:82] CH 11
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[Net-Creds] successfully stopped.
Thread::[Responder] successfully stopped.
Thread::[sslstrip] successfully stopped.

```