P0cL4bs/WiFi-Pumpkin-deprecated #164: no Internet Connection on Victim-device

theoneandonly-vector, 2017-01-04:
## What's the problem (or question)?
The Target-Device (Android Phone) doesn't have internet, when connected to PumpAP.

#### Please tell us details about your environment.
Wifi: Atheros Communications, Inc. AR9271 802.11n
+ Ethernet-connection
.
Wifi-Pumpkin: v0.8.4

* Virtual Machine (yes or no and which): no

Linux Debian 4.8.0-kali2-amd64 #1 SMP Debian 4.8.11-1kali1 (2016-12-08) x86_64 GNU/Linux

theoneandonly-vector, 2017-01-04:
Just did that, sadly it doesn't fix it.

"root@Debian:~# cd /opt/WiFi-Pumpkin/
root@Debian:/opt/WiFi-Pumpkin# git pull
remote: Counting objects: 29, done.
remote: Compressing objects: 100% (27/27), done.
remote: Total 29 (delta 9), reused 0 (delta 0), pack-reused 0
Unpacking objects: 100% (29/29), done.
From https://github.com/P0cL4bs/WiFi-Pumpkin
   5774392..de7eef0  master     -> origin/master
Updating 5774392..de7eef0
Fast-forward
 CHANGELOG                               |  2 ++
 README.md                               |  1 -
 core/config/commits/Lcommits.cfg        |  2 ++
 core/main.py                            | 10 ++++-----
 core/packets/dhcpserver.py              | 36 ++++++++++++++++++++++++++-------
 plugins/external/Responder/Responder.py |  1 +
 requirements.txt                        |  2 +-
 7 files changed, 39 insertions(+), 15 deletions(-)
root@Debian:/opt/WiFi-Pumpkin# git pull
Already up-to-date."

mh4x0f, 2017-01-04:
ok, I'm check this. wait...

mh4x0f, 2017-01-04:
@theoneandonly-vector change in Tab plugins for SSLstrip+/Dns2proxy and check internet on AP

theoneandonly-vector, 2017-01-04:
this works..

theoneandonly-vector, 2017-01-04:
but how could I now use pumpkin-proxy features, or BDFProxy-ng?

mh4x0f, 2017-01-04:
ok, wait I 'll fix this bug

mh4x0f, 2017-01-04:
I think this will solve the problem. try now and say if does work

theoneandonly-vector, 2017-01-04:
BDFProxy-ng and Pumpkin-Proxy:
 it's really weird.. It shows the "!"-mark beside the wifi-icon in android.
google, facebook and youtube do work. But I can't connect to other websites, http-connections wont work at all.

mh4x0f, 2017-01-04:
Which other websites?

theoneandonly-vector, 2017-01-04:
test.com
csploit.org
local.ch

theoneandonly-vector, 2017-01-04:
nsa.gov

mh4x0f, 2017-01-04:
ok, I check this and you are right. very strange. I'll check if the problem again.

mh4x0f, 2017-01-04:
No, brown i check now again. everything is alright. the problem is the server

closed 2017-01-04 by mh4x0f

theoneandonly-vector, 2017-01-04:
I just uninstalled, removed and redownloaded and reinstalled it.
still not fixed. 
test.com, putty.org works using SSLstrip+/Dns2proxy but doesn't when I use BDFProxy-ng or Pumpkin-Proxy

jonyvx, 2018-06-01:
hi..thanks for creating this cool software...but it dosnt share internet with victim !! it shows limited connection...no internet access...and if it works fine ...it works fine only for like 40 seconds ? and then no internet...access point works and can see it from devices but no internet on all plugins..and dosnt show how many clients are connected...always client 0 ......hope you fix it in future update...thanks

jonyvx, 2018-06-03:
edit: i found a solution for my problem ...wifipumpkin works good for me...internet sharing work ok ..but not all websites can be opened in other devices and https dosnt get striped from other devices and dosnt capture visited urls of google and many websites from uc browser and from android and from other browsers....can you add feature to limit speed of wifi and disconnect some clients ? and dns spoof dosnt change all website all the time...some times it redirect "all traffic" and some times it dosnt work...and wifipumpkin dosnt show clients connected numbers its shows (clients 0 ) when 2 devices connected...sslstrip work sometimes on sslstrip|proxy2dns ...and sslstrip and password and login credentials dosnt work on pumpkin proxy/tcp or in sergeo proxy /bgf/.....hope you fix them all in future update ..thanks

jonyvx, 2018-06-03:
..driftnet dosnt capture or dosnt show all visited websites...

jonyvx, 2018-06-03:
.driftnet dosnt capture or dosnt show all visited websites pictures only few pictures...

Rdgeno, 2019-09-13:
Has anyone figured out a definitive answer as to how to connect wifi pumpkin to the internet. I have tried all the solutions here and many others I can connect to the ap but still no internet.

ctrlheadhit, 2019-09-18:
Same problem i try every single option in every single comment.pls need help for make this powerfull tool working.

ctrlheadhit, 2019-09-18:
New problem...when i push start on the left down corner it stop loading after around 20%

scottharddick, 2019-10-31:
So sad. I have tried all these and many other fixes. This all looked so promising. I have not once gotten it to give internet access in every possible way there is. I spent at least 20 hours or more over two months on this. I am cutting my losses.