P0cL4bs/WiFi-Pumpkin-deprecated #165: Python DNS Server improvements

mh4x0f, 2017-01-06:
I need help for do test the new pyDNS Server implementation. the current version 0.8.4 is running the dns2proxy because i had some problems with DNS server and i was forced to solve this problem temporarily last commit. so i doing new test with new version the DNS server for add new features.

cbogithub, 2017-01-08:
DNS lookup failed : address 'wpad' not found 

[image]

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TP-link  model WN821N
* Version used tool: 0.8.4
* Virtual Machine (yes or no and which): WMware
* Operating System and version: Kali linux 2016.2

mh4x0f, 2017-02-16:
@TimSchumi Could you do some tests in your systems?

TimSchumi, 2017-02-16:
Of course

mh4x0f, 2017-02-16:
ok, I'll send the version .zip with pydns server implementation coming soon.

mh4x0f, 2017-02-25:
I add the version BETA test in Release tab, **wifi-pumpkin_0.8.4.2Beta_all.deb** I think it's best to use the same issue .

TimSchumi, 2017-02-25:
DNS Logger and Network Access work!

mh4x0f, 2017-02-25:
ok, did you try with which OS ?

TimSchumi, 2017-02-25:
Ubuntu 16.04 and Kali Linux 2016.2

mh4x0f, 2017-02-25:
nice, thank's brown. I'll add news features coming soon.

mh4x0f, 2017-02-25:
@TimSchumi Did you succeed install mitmproxy 0.17 normally on Kali Linux 2016.2. because a lot of people are reporting this bug.

TimSchumi, 2017-02-26:
I had to install it on the dirty way, not sure hiw I made it

mh4x0f, 2017-02-26:
ok, I'll download the Kali 2016.2 and make somes test.

closed 2017-05-04 by mh4x0f