P0cL4bs/WiFi-Pumpkin-deprecated #166: Error starting SSL server on port 443

cbogithub, 2017-01-06:
[image]
[image]
[image]

===============
Hello , I had written one article on this issue, and one that's someone said this is erorr on my system os, but i dont think so, .
- You can see second image . i test with deafault plug-in Responder in folder "/root/xxx/WiFi-Pumpkin/plugins/external". and it show error "[!] Error starting SSL server on port 443, check permissions or other servers running. 
"
- Then i tested Responder at https://github.com/SpiderLabs/Responder" (third image ) it working ...OMG

mh4x0f, 2017-01-06:
look this [issue](https://github.com/SpiderLabs/Responder/issues/70) The reason why the SSL server doesn't start is that SSLv3 is disabled in Debian's OpenSSL,  I think this bug has been fixed on last version the Responder. I'm use an old version the Responder. so I'll check this coming soon

mh4x0f, 2017-01-07:
now it possible bind port 443 with Responder. thanks for issue

closed 2017-01-07 by mh4x0f