P0cL4bs/WiFi-Pumpkin-deprecated #168: New Features

TimSchumi, 2017-01-06:
@mh4x0f 
I'm planning to add some features into [my fork](http://www.github.com/TimSchumi/WiFi-Pumpkin/tree/timschumi) of WiFi-Pumpkin and wanted to ask if you would like to merge some of them into the official program. If there are some, I would base them off the master branch for easy merging.

The current list of planned features is:
- WEP Support (maybe for pentesting training)
- Security Key check to make sure that the Key can be used (charset & length)

mh4x0f, 2017-01-06:
> WEP Support (maybe for pentesting training)

I think that WEP is dead, nobody use more. but add into fork i can see this, because if i should add this feature i would have to make other modifications for this I don't add the WEP support. 

> Security Key check to make sure that the Key can be used (charset & length)

again, I need to see how did you implement this feature.

TimSchumi, 2017-01-07:
@mh4x0f 
I now added the Key Checking (#169) and will wait if you merge it or not. Then I'll start to add WEP, because if you merge Key Checking I would add WEP Key support, too.