P0cL4bs/WiFi-Pumpkin-deprecated #174: support python3

blshkv, 2017-01-08:
It's getting more and more difficult to support tools which are python2.7 only. In particular, the latest version of mitmproxy is python3 only Please update.
Thank you.

mh4x0f, 2017-01-08:
I think that is not possible to implement, the project got too very bigger for change of that type. also, most plugin has been implemented python2.

blshkv, 2017-01-08:
It's never too late especially for little projects like yours.  Python 2 is the dead end which will happen  in 2020 (supposed to come in 2015 originally)

mh4x0f, 2017-01-09:
AFAIK, the python 2 will receive security updates until 2020 when its support will be discontinued. only this.

closed 2017-01-27 by mh4x0f

blshkv, 2017-05-18:
I still think you should change your mind and made an effort to migrate.

There are more and more dependences which you use support python3. Some of them, like mitmproxy (which is the main dep) does not even support python2 anymore.

blshkv, 2019-03-26:
https://pythonclock.org/