P0cL4bs/WiFi-Pumpkin-deprecated #181: Wifi-Pumpkin not working  on Parrot Sec Os  3.4.1

killheart1, 2017-01-30:
## What's the problem (or question)?

Hello Wifi-Pumpkin  Dev team :smiley:   I greatly appreciate the work and effort,  however
  Can't get it to run on  Pentesting linux operating system **Parrot Sec Os**...

I  always get this message::

WiFi-Pumpkin/settings/hostapd.conf
nl80211: Could not configure driver mode
nl80211: deinit ifname=wlan0 disabled_11b_rates=0
nl80211 driver initialization failed.
wlan0: interface state UNINITIALIZED->DISABLED
wlan0: AP-DISABLED 
hostapd_free_hapd_data: Interface wlan0 wasn't started

  I have spent much time trying to fix this with no succes and other software that uses the hostapd nl80211 driver works fine on Parrot Sec Os...

#### Please tell us details about your environment.
*
wireless card fully support AP mode 
(Qualcom Atheros QCA9565 / AR9565)
 driver=ath9k        driver version=4.8.0-parrot-amd64

* wifi-pumpkin works fine on kali linux using the same hardware configuration, whether it be in a VM or otherwise... just wont work on Parrot Sec Os

* Operating System and version: 
   Parrot Sec Os 3.4.1

killheart1, 2017-01-30:
Wifi-Pumpkin seems  to be working fine  on Parrot Sec Os 3.4.1  
 just need to manually  initialize wireless card for AP in monitor mode first..

closed 2017-01-30 by killheart1

mh4x0f, 2017-01-30:
I'll add fix for this bug coming soon

killheart1, 2017-01-30:
Awesome :smiley:

mh4x0f, 2017-01-31:
@killheart1 Which method did you use for set card in AP monitor? airmon-ng check kill ?

killheart1, 2017-02-01:
I did not use airmon-ng check kill ... (should i?)

the method  i used for set card  in  AP  was : airmon-ng start  (card-for-AP)
   example: airmon-ng start wlan0

[image]

Then i initialize Wifi-Pumpkin (or refresh network adapter selection menu if Wifi-Pumpkin was already initialized before putting card for AP in monitor mode)

[image]

Works great :smiley: