P0cL4bs/WiFi-Pumpkin-deprecated #182: No module named mitmproxy.models

Georgestrmina08, 2017-02-03:
## What's the problem (or question)?

Hello, I have a problem after command ......  sudo python ./wifi-pumpkin.py  ..... I'm getting this error :

sudo python ./wifi-pumpkin.py
Traceback (most recent call last):
  File "./wifi-pumpkin.py", line 45, in <module>
    from core.main import Initialize
  File "/home/eurolinux08/Desktop/WAY-FAY/WiFi-Pumpkin-master/core/main.py", line 31, in <module>
    from core.widgets.tabmodels import (
  File "/home/eurolinux08/Desktop/WAY-FAY/WiFi-Pumpkin-master/core/widgets/tabmodels.py", line 7, in <module>
    from core.utility.threads import ThreadPopen
  File "/home/eurolinux08/Desktop/WAY-FAY/WiFi-Pumpkin-master/core/utility/threads.py", line 19, in <module>
    from core.servers.proxy.controller.handler import MasterHandler
  File "/home/eurolinux08/Desktop/WAY-FAY/WiFi-Pumpkin-master/core/servers/proxy/controller/handler.py", line 1, in <module>
    from plugins.extension import *
  File "/home/eurolinux08/Desktop/WAY-FAY/WiFi-Pumpkin-master/plugins/extension/beef.py", line 2, in <module>
    from mitmproxy.models import decoded
ImportError: No module named mitmproxy.models

I  couldn't solve problem.Please help me?

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter 
* Version used tool: latest one -  wifi-pumpkin-0.8.4
* Virtual Machine (yes or no and which): NO 
[image]

* Operating System and version: 
Linux 4.4.0-59-generic x86_64

TimSchumi, 2017-02-03:
Try running "pip install mitmproxy==0.17" manually

Georgestrmina08, 2017-02-03:
I try this command but I'm getting this now: 
[image]

Georgestrmina08, 2017-02-03:
I did and I repeat process ones again but the same problem is coming.
[image]

TimSchumi, 2017-02-03:
Maybe you need to update pip first.

Also, do you have a good (and stable) connection?

Georgestrmina08, 2017-02-03:
Version of my pip is : pip 9.0.1 from /home/eurolinux08/.local/lib/python2.7/site-packages (python 2.7)
My connection is stable and fast .

TimSchumi, 2017-02-03:
Then it is a problem with your specific installation

Georgestrmina08, 2017-02-04:
Do you have some idea how I can fix this problem? I will apriciate ,thanks

closed 2017-02-06 by Georgestrmina08