P0cL4bs/WiFi-Pumpkin-deprecated #184: ERROR Hostpad

Georgestrmina08, 2017-02-06:
## What's the problem (or question)?
I install wifi-pumpkin everything was fine and now when I start the application and go to start I'm getting this message.

[image]

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): 
COMMAND: airmon-ng 
PHY	Interface	Driver		Chipset

phy0	wlp2s0		ath9k		Qualcomm Atheros QCA9565 / AR9565 Wireless Network Adapter (rev 01)
phy1	wlx00c0ca91ec8c	rt2800usb	Ralink Technology, Corp. RT3572

COMMAND : ifconfig
enp1s0    Link encap:Ethernet  HWaddr 30:65:ec:99:cb:fb  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

lo        Link encap:Local Loopback  
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:687 errors:0 dropped:0 overruns:0 frame:0
          TX packets:687 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1 
          RX bytes:108022 (108.0 KB)  TX bytes:108022 (108.0 KB)

wlp2s0    Link encap:Ethernet  HWaddr 44:1c:a8:0a:d4:fb  
          inet addr:192.168.1.39  Bcast:192.168.1.255  Mask:255.255.255.0
          inet6 addr: fe80::2ee2:7f23:82ef:17fb/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:17755 errors:0 dropped:0 overruns:0 frame:0
          TX packets:10854 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:20868965 (20.8 MB)  TX bytes:1877282 (1.8 MB)

wlx00c0ca91ec8c Link encap:Ethernet  HWaddr 00:c0:ca:91:ec:8c  
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

 Version used tool: new
* Virtual Machine (yes or no and which): no
* Operating System and version: Linux Mint 18.1 Serena

please, someone, help me I was doing research on internet and many people have the same problem but no solution.

TimSchumi, 2017-02-07:
Try running this command in Terminal:
sudo nmcli radio wifi off
If that fails, try this instead:
sudo nmcli nm wifi off

Then WiFi Pumpkin should start

To enable WiFi for your Computer again, either reboot your system or run the command you also ran above but replace off with on. After that, you may need to restart the NetworkManager service

Georgestrmina08, 2017-02-07:
I tried this steps but nothing again I'm getting same problem:

[ERROR Hostapd] 

Output::
nl80211 driver initialization failed.
wlx00c0ca91 ec8c: interface state UNINITALIZED->DISABLED
wlx00c0ca91 ec8c : AP-DISABLED
hostapd_free_hapd_data:Interface wlx00c0ca91 ec8c wasnt started

Please help .

TimSchumi, 2017-02-07:
Do you know if your System uses NetworkManager?

Georgestrmina08, 2017-02-07:
No, I don't know please explain me, when I'm doing your first step ----- sudo nmcli radio wifi off ---- I'm not getting any error but the problem is I'm losing internet and then wifi pumpkin is not allowing me to start the process because they asking for the internet connection.

closed 2017-02-09 by Georgestrmina08

Bee1To, 2017-02-09:
Try wpa_supplicant for wifi without Network-manager being on.

TimSchumi, 2017-02-21:
@Georgestrmina08
You might try installing the aircrack-ng suite and run
"(sudo) aircrack-ng check kill"
This should kill off interfering processes, but make sure you connect to your WiFi first