P0cL4bs/WiFi-Pumpkin-deprecated #185: gui of wifi-pumpkin not visible

Treco-one, 2017-02-07:
## What's the problem (or question)?

i m not able to see anyting for except the wifi-pumpkin window n that too with a black screen with nothing visible

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: wifi-pumpkin-0.8.3 it happens on every version
* Virtual Machine (yes or no and which): no dual boot kali and windows 7
* Operating System and version: kali 2.2016 rolling 
[image]

TimSchumi, 2017-02-07:
Uninstall Version 0.8.3 and install 0.8.4 through the installer script. There may be some missing dependencies

Treco-one, 2017-02-07:
I'll  try this one, 
but I've tried most of the previous versions and it met the same problem 
Still I'll try 
Regards

Treco-one, 2017-02-08:
i installed 0.8.4 through the installer script but it showed the same error
Loading GUI...
WiFi-Pumpkin Running!
X Error: BadAccess (attempt to access private resource denied) 10
  Extension:    130 (MIT-SHM)
  Minor opcode: 1 (X_ShmAttach)
  Resource id:  0x141
X Error: BadShmSeg (invalid shared segment parameter) 128
  Extension:    130 (MIT-SHM)
  Minor opcode: 5 (X_ShmCreatePixmap)
  Resource id:  0x1000019
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x100001a
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x100001a
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x100001a
X Error: BadDrawable (invalid Pixmap or Window parameter) 9
  Major opcode: 62 (X_CopyArea)
  Resource id:  0x100001a
i tried the fix shown in 
BadDrawable (invalid Pixmap or Window parameter) #53
by adding 
export QT_X11_NO_MITSHM=1 in 
~/.profile file
but still the error remains the same

Treco-one, 2017-02-09:
Help me guyz

TimSchumi, 2017-02-09:
Did you try this: http://www.stackoverflow.com/questions/18429484/pyqt4-x-errors-when-starting-aplication
?

Treco-one, 2017-02-09:
Thanks for the reply,   
I tried it but i didn't helped 
Is there any other way?  Help me out guyz

TimSchumi, 2017-02-09:
Then it's a specific problem with your setup

mh4x0f, 2017-02-09:
run the app without gksudo command. execute "sudo wifi-pumpkin".

Treco-one, 2017-02-11:
With sudo i can see the above error as u can see in the screenshot I've mentioned 
But wether i use gksudo or sudo the interface doesn't seems to work
Regards

godrockz, 2017-02-15:
i had the same issues with the vncserver. solved with(not recommend): 
[root@alarmpi alarm]# /usr/bin/vncserver -geometry 1440x900 -alwaysshared -fg
after login: xhost + and now start wifi-pumpkin

note: QT can't elevate permission to access some shared memory, because X is running with user privileges - start X as root and it will work

closed 2017-02-26 by mh4x0f

SaulAgui, 2017-05-16:
getting the exact same error, tryed all the same options. plz help

becare, 2017-06-16:
sudo nano /etc/environment

Add this line
QT_X11_NO_MITSHM=1

Enjoy!

ArnoldSkylake, 2019-04-05:
> sudo nano /etc/environment
> 
> Add this line
> QT_X11_NO_MITSHM=1
> 
> Enjoy!

BRO, THIS HELP ME !!!!
Thaaaaanks

u666sa, 2019-11-15:
> sudo nano /etc/environment
> 
> Add this line
> QT_X11_NO_MITSHM=1
> 
> Enjoy!

Thank you so much for this one! Fedora 30 is running pumpkin!

Arturo-Melgarejo, 2020-01-05:
yoooo thanks this fix helps!! i have try to run in ubuntu/kali 64 and it works:)