P0cL4bs/WiFi-Pumpkin-deprecated #187: there's no clients on the home tab

LeapSpace, 2017-02-17:
## What's the problem (or question)?

*there's no clients on the home tab, but in fact,my cellphone has connected*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: v0.8.4
* Virtual Machine (yes or no and which): yes
* Operating System and version: 

Linux bogon 4.6.0-kali1-amd64 #1 SMP Debian 4.6.4-1kali1 (2016-07-21) x86_64 GNU/Linux

[image]

TimSchumi, 2017-02-17:
Because there are no clients connected. Look at the bottom right corner

LeapSpace, 2017-02-17:
not really.
My cellphone has connected, and you can see there's data stream,but clients count still be zero.
[image]

mh4x0f, 2017-02-18:
@LeapSpace did you try connect another device on AP?  
try this:  change the **python DHCP Server** to **isc DHCPD Server** on Settings tab as can see on screenshot.
[image]

LeapSpace, 2017-02-20:
@mh4x0f 
I didn't connect another device. but when I  change the option python DHCP Server to isc DHCPD Server according to what you said,it seems ok. is there something wrong with the python DHCP Server?

mh4x0f, 2017-02-22:
@LeapSpace yes,the tool isc dhcp server is more efficient for get info from device.

closed 2017-02-22 by mh4x0f

hackituria, 2018-10-04:
I'am using Wifi-Pumpkin v0.8.5 and Kali Linux 2018.4.

In my case, some smartphones are displayed on Home tab but other not. However, analyzing reported logs I can see that the smartphone was detected and connected to Rogue AP.

```
Session information::

[*] ESSID AP: 
[*] AP Create at: Thu Oct  4 12:09:27 2018
[*] AP Down   at: Thu Oct  4 12:12:26 2018

Report Generated at::Thu Oct  4 13:23:17 2018

-[ hostapd Logger ]-
SessionID[MTAyNTA=] 2018-10-04 12:09:34,035 : Configuration file: core/config/hostapd/hostapd.conf
SessionID[MTAyNTA=] 2018-10-04 12:11:16,587 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: authenticated
SessionID[MTAyNTA=] 2018-10-04 12:11:16,595 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: associated (aid 1)
SessionID[MTAyNTA=] 2018-10-04 12:11:16,595 : wlan0: AP-STA-CONNECTED 10:b1:f8:ae:c0:4b
SessionID[MTAyNTA=] 2018-10-04 12:11:16,596 : wlan0: STA 10:b1:f8:ae:c0:4b RADIUS: starting accounting session D85D2D32B1793B62
SessionID[MTAyNTA=] 2018-10-04 12:11:18,592 : wlan0: AP-STA-DISCONNECTED 10:b1:f8:ae:c0:4b
SessionID[MTAyNTA=] 2018-10-04 12:11:19,594 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
SessionID[MTAyNTA=] 2018-10-04 12:11:23,136 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: authenticated
SessionID[MTAyNTA=] 2018-10-04 12:11:23,144 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: associated (aid 1)
SessionID[MTAyNTA=] 2018-10-04 12:11:23,144 : wlan0: AP-STA-CONNECTED 10:b1:f8:ae:c0:4b
SessionID[MTAyNTA=] 2018-10-04 12:11:23,144 : wlan0: STA 10:b1:f8:ae:c0:4b RADIUS: starting accounting session 87C8426CB6DB3A87
SessionID[MTAyNTA=] 2018-10-04 12:11:44,679 : wlan0: AP-STA-DISCONNECTED 10:b1:f8:ae:c0:4b
SessionID[MTAyNTA=] 2018-10-04 12:11:45,691 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
SessionID[MTAyNTA=] 2018-10-04 12:11:48,422 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: authenticated
SessionID[MTAyNTA=] 2018-10-04 12:11:48,429 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: associated (aid 1)
SessionID[MTAyNTA=] 2018-10-04 12:21:57,425 : Configuration file: core/config/hostapd/hostapd.conf
SessionID[MTAyNTA=] 2018-10-04 12:23:31,758 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: authenticated
SessionID[MTAyNTA=] 2018-10-04 12:23:31,771 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: associated (aid 1)
SessionID[MTAyNTA=] 2018-10-04 12:23:34,174 : wlan0: AP-STA-DISCONNECTED 10:b1:f8:ae:c0:4b
SessionID[MTAyNTA=] 2018-10-04 12:23:35,181 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
SessionID[MTAyNTA=] 2018-10-04 12:23:43,634 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: authenticated
SessionID[MTAyNTA=] 2018-10-04 12:23:43,642 : wlan0: STA 10:b1:f8:ae:c0:4b IEEE 802.11: associated (aid 1)
```

¿How can I fix it?