P0cL4bs/WiFi-Pumpkin-deprecated #189: wifi pumpkin startup error

jeretc, 2017-02-20:
## What's the problem (or question)?

*Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [MjEyNTk=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 5787 (hostapd)]
[New Thread 5788 (DNSServer)]
[New Thread 5793 (Net-Creds)]
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1433, in Start_PumpAP
    thread.start()
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 265, in start
    config = proxy.ProxyConfig(port=8080,mode='transparent')
TypeError: __init__() got an unexpected keyword argument 'mode'
*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TL-722N
* Version used tool: 0.8.4
* Virtual Machine (yes or no and which): No
* Operating System and version: parrot 3.5

LeapSpace, 2017-02-21:
it needs mitmproxy 0.17
try sudo pip install mitmproxy==0.17

jeretc, 2017-02-21:
hi, i tried installing mitmproxy, but the installation is failing.
[mitmproxy.txt](https://github.com/P0cL4bs/WiFi-Pumpkin/files/788967/mitmproxy.txt)

jeretc, 2017-02-22:
installation is failing.

  Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-pnjKsp/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-W84WuS-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-pnjKsp/cryptography/
----------------------------------------

jeretc, 2017-02-23:
build/temp.linux-x86_64-2.7/_openssl.c:70047:29: note: (near initialization for ‘_cffi_struct_unions[133].alignment’)
  error: command 'x86_64-linux-gnu-gcc' failed with exit status 1

closed 2017-02-26 by mh4x0f

mh4x0f, 2017-02-26:
#196 try this