P0cL4bs/WiFi-Pumpkin-deprecated #190: Use of external wifi usb adapter

shivgori, 2017-02-21:
## What's the problem (or question)?

*I want to use wlan0 (internal card) to connect to Internet (with my phone as a Wi-Fi Hotspot) and a second physical Wifi adapter (tplink WN722N) to create the fake AP and make all stuff provided by wifi-pumpkin.*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):tplink WN722N
* Version used tool: 0.8.4
* Virtual Machine (yes or no and which): no
* Operating System and version: ubuntu 16.10

TimSchumi, 2017-02-21:
This is already built into the Program.
In the Settings Dialog, there should be a drop-down box where you can select your Adapter

shivgori, 2017-02-21:
it show this error
[image]

TimSchumi, 2017-02-21:
You might try installing the aircrack-ng suite and run
"(sudo) airmon-ng check kill"
But make sure you have connected your wlan0 to your WiFi-Hotspot because this command will probably kill NetworkManager

closed 2017-02-22 by mh4x0f

syira7, 2018-11-19:
do you need external wireless adapter for you to use Wifi pumpkin?