P0cL4bs/WiFi-Pumpkin-deprecated #191: KeyError: 'Mac Address'

TimSchumi, 2017-02-21:
## What's the problem (or question)?

Getting this traceback in the log. It does not seem to break anything, but it looks kind of suspicious:

Traceback (most recent call last):
  File "/home/timschumi/Schreibtisch/WiFi-Pumpkin/core/main.py", line 926, in GetDHCPDiscoverInfo
    self.Add_data_into_QTableWidget(self.APclients)
  File "/home/timschumi/Schreibtisch/WiFi-Pumpkin/core/main.py", line 905, in Add_data_into_QTableWidget
    self.THeaders['Mac Address'].append(mac)
KeyError: 'Mac Address'

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TP-LINK WN-722N
* Version used tool: 0.8.4
* Virtual Machine (yes or no and which): No
* Operating System and version: Ubuntu 16.04

mh4x0f, 2017-02-23:
I don't find any problem in this lines. btw, this is happening for all device that try connect with AP ?

TimSchumi, 2017-02-23:
Yes, this is happening when a device was connected to the AP. I only tested with a Nintendo 3DS, but I can test with more devices and if it only happens on first connect

closed 2017-03-07 by mh4x0f