P0cL4bs/WiFi-Pumpkin-deprecated #193: Wifi-pumpkin does not start completely

Bee1To, 2017-02-24:
## What's the problem (or question)?
Wifi-pumpkin does not start completely and cannot detect when I connect to ap with cellphone I have kali 2.0 rolling and using awus036nh usb adapter for wlan1 as fake ap and eth0 for internet access 
*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

Bee1To, 2017-02-24:
No vm, kali rolling 2016.1, wifi-pumpkin version  0.8.4

Bee1To, 2017-02-24:
[image]

mh4x0f, 2017-02-24:
it needs mitmproxy 0.17
``` sh
sudo pip install mitmproxy==0.17
```

closed 2017-02-27 by mh4x0f