P0cL4bs/WiFi-Pumpkin-deprecated #195: update version mitmproxy 0.17 to 0.18.2

mh4x0f, 2017-02-26:
I'm receive a lot report bug the version **mitmproxy 0.17** is default on **WP 0.8.4**. because of this, I 'll update for version 0.18.2 but for make this i need to do some test in different OS based in Debian (**Kali Linux 2016.2**,Parrot 3.4 e etc). OBS: I already did the implementation the code for support mitmproxy 18.2.

jeretc, 2017-02-27:
└──╼ $sudo pip install mitmproxy==0.18.2
[sudo] password for parrot: 
Requirement already satisfied: mitmproxy==0.18.2 in /usr/lib/python2.7/dist-packages
Requirement already satisfied: enum34<2,>=1.0.4 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: ipaddress<1.1,>=1.0.15 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: typing==3.5.2.2 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2)

Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1433, in Start_PumpAP
    thread.start()
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 265, in start
    config = proxy.ProxyConfig(port=8080,mode='transparent')
TypeError: __init__() got an unexpected keyword argument 'mode'

mh4x0f, 2017-02-27:
- mitmproxy 0.18.2 installation 

| OS | Description |
| --- | --- |
| Parrot OS 3.4.1 | supported |
| Kali Linux 2016.2  | not tested |

mh4x0f, 2017-03-03:
@TimSchumi could you try install mitmproxy 0.18.2 on kali 2016.2?

shivgori, 2017-03-04:
after update it gives error
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 45, in <module>
    from core.main import Initialize
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 31, in <module>
    from core.widgets.tabmodels import (
  File "/usr/share/WiFi-Pumpkin/core/widgets/tabmodels.py", line 7, in <module>
    from core.utility.threads import ThreadPopen
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 19, in <module>
    from core.servers.proxy.controller.handler import MasterHandler
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/controller/handler.py", line 43, in <module>
    class MasterHandler(flow.FlowMaster):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/controller/handler.py", line 87, in MasterHandler
    @controller.handler
AttributeError: 'module' object has no attribute 'handler'

mh4x0f, 2017-03-04:
@shivgori Update mitmproxy to 0.18.2.
``` sh
sudo pip install mitmproxy==0.18.2 
```

jeretc, 2017-03-04:
@mh4x0f re-installed it working fine now. thanks

shivgori, 2017-03-04:
thanks after reinstalled it works perfect

closed 2017-03-08 by mh4x0f

BarAluf, 2018-07-15:
hi everyone.
i want to upgrade mitm from version 2.0.2 to version 4.
this is the error module 'mitmproxy.controller' has no attribute 'handler'.
how can i easily convert this.