P0cL4bs/WiFi-Pumpkin-deprecated #196: MITMPROXY and PumpkinProxy Error

PhantomXMT, 2017-02-26:
## What's the problem (or question)?

when i installed wifi pumpkin i got this error:
  Failed building wheel for cryptography

  Failed building wheel for pillow

and

Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-Rc6x83/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-gqHQ93-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-Rc6x83/cryptography/

and when i start the program it says on Pumpkin Proxy it says:
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1433, in Start_PumpAP
    thread.start()
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 265, in start
    config = proxy.ProxyConfig(port=8080,mode='transparent')
TypeError: __init__() got an unexpected keyword argument 'mode'

Thank you for your help.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):TP-LINK TL-WN722N
* Version used tool: V0.8.4
* Virtual Machine (yes or no and which): No 
* Operating System and version: Parrot Security OS 3.4.1

mh4x0f, 2017-02-26:
try this:
```sh 
sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev
```
and now install mitmproxy.
``` sh 
sudo pip install mitmproxy==0.17
```

PhantomXMT, 2017-02-27:
Tried the first command and this comes up:
sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev
Reading package lists... Done
Building dependency tree       
Reading state information... Done
libffi-dev is already the newest version (3.2.1-6).
libssl-dev is already the newest version (1.1.0e-1).
libxml2-dev is already the newest version (2.9.4+dfsg1-2.2).
libxslt1-dev is already the newest version (1.1.29-2).
python-dev is already the newest version (2.7.13-2).
python-pip is already the newest version (9.0.1-2).
0 upgraded, 0 newly installed, 0 to remove and 1235 not upgraded.

and i still have the same error.

jeretc, 2017-02-27:
yes, same here.

mh4x0f, 2017-02-27:
try install the version 0.18.2 the mitmproxy.
``` sh
sudo pip install mitmproxy==0.18.2
```
if work fine I 'll add the version with mitmproxy updated.

jeretc, 2017-02-27:
$sudo pip install mitmproxy==0.18.2
[sudo] password for parrot: 
Requirement already satisfied: mitmproxy==0.18.2 in /usr/lib/python2.7/dist-packages
Requirement already satisfied: enum34<2,>=1.0.4 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: ipaddress<1.1,>=1.0.15 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2)
Requirement already satisfied: typing==3.5.2.2 in /usr/lib/python2.7/dist-packages (from mitmproxy==0.18.2)

└──╼ $sudo wifi-pumpkin
[sudo] password for parrot: 
Loading GUI...
WiFi-Pumpkin Running!
libpng warning: iCCP: known incorrect sRGB profile

[*] Loading debugging mode
[*] Current Session::ID [OTU4OTA=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 2788 (hostapd)]
[New Thread 2789 (DNSServer)]
[New Thread 2794 (Net-Creds)]
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1433, in Start_PumpAP
    thread.start()
  File "/usr/share/WiFi-Pumpkin/core/utility/threads.py", line 265, in start
    config = proxy.ProxyConfig(port=8080,mode='transparent')
TypeError: __init__() got an unexpected keyword argument 'mode'

mh4x0f, 2017-02-27:
OK, Parrot OS already support mitmproxy 0.18.2 by default. now I need to try install this module in Kali Linux 2016.2 and if work fine I'll update this.

jeretc, 2017-02-27:
but the error still remains.

mh4x0f, 2017-02-27:
@jeretc Logic, as I said above I will still add the new version with support for mitmproxy 0.18.2 wait

jeretc, 2017-02-27:
right ..

nedege, 2017-02-28:
problem pupkin-proxy not working in kali-linux-xfce-W07-amd64
removed in requierments.txt
mitmproxy==0.17
and install wifi-pumpkin 
remove mitmproxy from kali
apt-get autoremove mitmproxy
install this
apt-get install libpq-dev python-dev libxml2-dev libxslt1-dev libldap2-dev libsasl2-dev libffi-dev libtiff5-dev libjpeg62-turbo-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev tcl8.6-dev tk8.6-dev python-tk
download from https://github.com/mitmproxy/mitmproxy/releases
mitmproxy 0.17 unzip and install
this will fail but dont worry 
reinstall mitmproxy
apt-get install mitmproxy
now pupkin proxy is working

why i don`t know

PhantomXMT, 2017-02-28:
I fixed the problem by doing:

sudo apt-get install python3-dev python3-pip libffi-dev libssl-dev
sudo pip3 install mitmproxy  # or pip3 install -- user mitmproxy

which installed mitmproxy v2.0 and it works :)

nedege, 2017-02-28:
verified working on kali-linux-xfce-2016.2-amd64
still have removed 
mitmproxy==0.17
thanks

jeretc, 2017-03-01:
@PhantomXMT i get this error

sudo pip3 install mitmproxy

Command "python setup.py egg_info" failed with error code -11 in /tmp/pip-build-q0ul_530/pyperclip/

PhantomXMT, 2017-03-01:
Well idk for me it worked

mh4x0f, 2017-03-03:
#195 solved now.

closed 2017-03-03 by mh4x0f

fir0j, 2018-12-08:
> try install the version 0.18.2 the mitmproxy.
> 
> ```shell
> sudo pip install mitmproxy==0.18.2
> ```
> 
> if work fine I 'll add the version with mitmproxy updated.

Thanks alot you guys!!! IT WORKED
1. sudo apt install python3-pip
2. sudo apt-get install python-pip python-dev libffi-dev libssl-dev libxml2-dev libxslt1-dev
3. sudo pip install mitmproxy==0.18.2