P0cL4bs/WiFi-Pumpkin-deprecated #199: DNS Spoofer Page Redirection

jeretc, 2017-03-06:
## What's the problem (or question)?

*Just checking if the dns spoofer page redirection still working? i'm running the phishing manager and dns spoofer but i can't seem to redirect non-https sites to a fake page.*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TP_722N
* Version used tool:  8.4
* Virtual Machine (yes or no and which): no
* Operating System and version: parrotsec 3.5

TimSchumi, 2017-03-06:
The Device not showing up is a problem of the DHCP Server. Try switching it in the Settings

mh4x0f, 2017-03-06:
>  i can't seem to redirect non-https sites to a fake page

[image]

closed 2017-03-06 by mh4x0f

jeretc, 2017-03-06:
hmmm.. not sure what it's suppose to mean tho i'm a fan of star wars.

TimSchumi, 2017-03-06:
In the move he screams "Nooooooooooooooo"

jeretc, 2017-03-06:
yesssss!!!

jeretc, 2017-03-06:
@TimSchumi thanks. it solved both issues.

TimSchumi, 2017-03-06:
I don't really know if I added valuable information to solve this but... Ok

shivgori, 2017-03-08:
hey jeretc, i am also facing same problem how it can be solve

jeretc, 2017-03-08:
Change the dhcp server under the settings.

shivgori, 2017-03-08:
what should i change

shivgori, 2017-03-08:
after select python dhcp server, connected device do not show in home

jeretc, 2017-03-08:
the dhcp server options, you can select a different dhcp server.

jeretc, 2017-03-08:
choose another not the python dhcp server. It should work.

shivgori, 2017-03-08:
k it's work. But only http sites redirect on fake page. HTTPS sites not open

jeretc, 2017-03-08:
of course, it won't work on https because all modern web browsers has HSTS.

tuthieu, 2017-03-25:
> jeretc
Is there anyway to defeat HSTS of browser?
Wifi-Pumpkin dev said "added new bypass HSTS by LeonardoNve sslstrip feature"

jeretc, 2017-03-25:
I have not tried that. You can get it to work with bettercap and
wifi-pumpkin.

tuthieu, 2017-03-25:
Hello jeretc,I don't know how to redirect with DNSSpoof.
First,I start AP,then start Phising and DNSSpoof.I check Redirect for all sites,but only when I go  10.0.0.1,the page show the index.html.When I go google.comor all other sitse the page doesn't return.

tuthieu, 2017-03-25:
I want to use evil twin(fluxion with SSL)to hack wifi,but Chrome has HSTS.
So I'm trying to make Wifi-Pumpkin to redirect to my evil twin page because Wifi-Pumpkin can defeat HSTS

jeretc, 2017-03-25:
It should work for google.com. Try changing the dhcp server under settings.

tuthieu, 2017-03-26:
Thanks jeretc!

tuthieu, 2017-05-20:
Hello admins,
When the victim go to HTTPS website,the redirection does not work.So I want Dnsspoofer redirect only HTTP and when victim go to HTTPS website,victim can still access that website.
Can you help me config Wifi-Pumpkin and Dnsspoofer to redirect with only HTTP?
Thanks you so much!