P0cL4bs/WiFi-Pumpkin-deprecated #207: Hint I update the software after the installation is complete, updated open like this again

wirelessice, 2017-03-30:
## What's the problem (or question)?
[image]
   @controller.handler
AttributeError: 'module' object has no attribute 'handler'

*answer here*
ubuntu 14.04

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
* Version used tool: 
* Virtual Machine (yes or no and which): 
* Operating System and version:

wirelessice, 2017-03-30:
Service_identity has been installed

mh4x0f, 2017-04-13:
solution: https://github.com/P0cL4bs/WiFi-Pumpkin/issues/195#issuecomment-284147071

closed 2017-04-13 by mh4x0f