P0cL4bs/WiFi-Pumpkin-deprecated #212: after device connected in home menu no device show

shivgori, 2017-04-18:
## What's the problem (or question)?

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): tp lnk722n
* Version used tool: latest
* Virtual Machine (yes or no and which): no
* Operating System and version: ubuntu

mh4x0f, 2017-04-21:
if you try with dhcp server and doesn't work, probably the problem is you device. did you try connect another device on AP?

shivgori, 2017-04-21:
yes i also used another device but problem not solve

mh4x0f, 2017-04-24:
ok, try this: 
using the options dhcp-server start PumpAP, connect the device and execute this command on terminal:
``` sh
cat /var/lib/dhcp/dhcpd.leases
```
send-me the output

closed 2017-04-27 by mh4x0f

shivgori, 2017-04-28:
I uninstall first and again install this tool and now it's working fine.

sorry for late reply