P0cL4bs/WiFi-Pumpkin-deprecated #213: SSID not visible

raulpresa, 2017-04-28:
## What's the problem (or question)?

USB card is ok, and support AP, I run wifi-pumpkin but cant find the SSID in the list of available access points

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):ENCORE (USB) N150 adapter.  Supports AP/mode
* Version used tool: 0.8.4
* Virtual Machine (yes or no and which): yes, KALI 2017
* Operating System and version: kali 2017

mh4x0f, 2017-04-28:
> Virtual Machine (yes or no and which): yes, KALI 2017

do you using which software for create Virtual Machine ?

raulpresa, 2017-04-28:
Oracle Virtual box

mh4x0f, 2017-04-28:
maybe, this is the problem "Oracle Virtual box". I had some bugs this type when I use this software and  I don't know how to solve this problem. I recommend that you use VMware.

raulpresa, 2017-04-28:
Ok. I'll try with a fresh ubuntu vm and if it fail, I'll try VMware.
Thank you for your help
Kind regards,

closed 2017-04-30 by mh4x0f