P0cL4bs/WiFi-Pumpkin-deprecated #217: OSError: [Errno 2] No such file or directory?

Mas0x, 2017-05-10:
root@kali:~# wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [MTIwMTg=]
[*] Configuring hostapd...
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 1329, in Start_PumpAP
    set_monitor_mode(self.SettingsEnable['AP_iface']).setDisable()
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 54, in setDisable
    Popen(['ifconfig', self.interface, 'down'])
  File "/usr/lib/python2.7/subprocess.py", line 390, in __init__
    errread, errwrite)
  File "/usr/lib/python2.7/subprocess.py", line 1024, in _execute_child
    raise child_exception
OSError: [Errno 2] No such file or directory

Can you assist?

mh4x0f, 2017-05-10:
hi, you're right, the problem is your adapter, this chipset does not support monitor mode. but I want to check this. send me output the command bellow:
``` sh
iw list
```

Mas0x, 2017-05-11:
Well this is interesting...

root@kali:~# iw list | grep "Supported interface modes" -A 10
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * monitor
		 * P2P-client
		 * P2P-GO
	Band 1:
		Capabilities: 0x1862
			HT20/HT40
			Static SM Power Save

Mas0x, 2017-05-11:
When using Mana-toolkit I egt this in verbose...

hostname WRT54G
Current MAC:   60:e3:27:0a:9d:ac (TP-LINK TECHNOLOGIES CO.,LTD.)
Permanent MAC: 60:e3:27:0a:9d:ac (TP-LINK TECHNOLOGIES CO.,LTD.)
New MAC:       b2:22:e3:d3:55:00 (unknown)
Network driver didn't actually change to the new MAC!!
Configuration file: /etc/mana-toolkit/hostapd-mana.conf
Using interface wlan0 with hwaddr 60:e3:27:0a:9d:ac and ssid "TalkTalk118"
wlan0: interface state UNINITIALIZED->ENABLED
wlan0: AP-ENABLED

mh4x0f, 2017-05-12:
> I can connect to the AP using the same wifi dongle that I use with WiFi-Pumpkin?

yes, if the Mana-toolkit work fine. try this, you need to remove or comment this [line](https://github.com/P0cL4bs/WiFi-Pumpkin/blob/master/core/main.py#L1329) number**1329** from file **WiFi-Pumpkin/core/main.py**. and try start AP.

Mas0x, 2017-05-12:
#set_monitor_mode(self.SettingsEnable['AP_iface']).setDisable() <<-- line 1329?

mh4x0f, 2017-05-12:
I need to check if your system is installed the ifconfig tool.
send me output the command bellow:
``` sh
ifconfig
```

Mas0x, 2017-05-12:
root@kali:~# ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.46  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ba27:ebff:fe8f:9209  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8f:92:09  txqueuelen 1000  (Ethernet)
        RX packets 35080  bytes 19563501 (18.6 MiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 38662  bytes 48699013 (46.4 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        ether 60:e3:27:0a:9d:ac  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

mh4x0f, 2017-05-12:
wait.. I'll add a simple code to check the bug.

mh4x0f, 2017-05-12:
@Mas0x  create the file test.py with this code bellow and run **"sudo python test.py"** in your system. I want to check is the problem is command split from string. after that, send me the output. 
``` python 

import shlex
import os
from subprocess import (
    Popen,PIPE,call,check_output,
)

interface = 'wlan0'

subnet='10.0.0.0'
router='10.0.0.1'
netmask='255.0.0.0'

SettingsAP = {
'iface':
    [
        'ifconfig %s up'%(interface),
        'ifconfig %s %s netmask %s'%(interface,router,netmask),
        'ifconfig %s mtu 1400'%(interface),
        'route add -net %s netmask %s gw %s'%(subnet,netmask,router)
    ],
'kill':
    [
        'iptables --flush',
        'iptables --table nat --flush',
        'iptables --delete-chain',
        'iptables --table nat --delete-chain',
        'ifconfig %s 0'%(interface),
        'killall dhpcd',
    ]}

for item in SettingsAP['iface']:
	Popen(shlex.split(item), stdout=PIPE,shell=False,stderr=PIPE)
os.system('ifconfig')
print("\nrestore settings...")
for item in SettingsAP['kill']:
	Popen(shlex.split(item), stdout=PIPE,shell=False,stderr=PIPE)
```

Mas0x, 2017-05-12:
root@kali:~# python test.py
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.46  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ba27:ebff:fe8f:9209  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8f:92:09  txqueuelen 1000  (Ethernet)
        RX packets 6684  bytes 312860 (305.5 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 12790  bytes 18555055 (17.6 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

restore settings...

Mas0x, 2017-05-12:
I've been testing today with WiFi-Pumpkin and Mana-toolkit....

Mana-toolkit basically switches the WiFi adapter on... (Master mode)

mh4x0f, 2017-05-12:
do you card adapter is connected ?

Mas0x, 2017-05-12:
always connected however Kali does not recognize any driver? and therefore does not see it in ifconfig, BUT does see it when using iwconfig.
root@kali:~# iwconfig
wlan0     unassociated  Nickname:"<WIFI@REALTEK>"
          Mode:Auto  Frequency=2.412 GHz  Access Point: Not-Associated   
          Sensitivity:0/0  
          Retry:off   RTS thr:off   Fragment thr:off
          Encryption key:off
          Power Management:off
          Link Quality:0  Signal level:0  Noise level:0
          Rx invalid nwid:0  Rx invalid crypt:0  Rx invalid frag:0
          Tx excessive retries:0  Invalid misc:0   Missed beacon:0

lo        no wireless extensions.

eth0      no wireless extensions.

Mas0x, 2017-05-12:
So when using Mana it WILL create an AP which I can gain access to and has internet connectivity, which is confusing as there is no recognized driver for it?

mh4x0f, 2017-05-12:
Disconnect the WIFi@REAETECK and run the script above  and change the line "os.system('ifconfig')" to os.system('ifconfig -a')

Mas0x, 2017-05-12:
Same output as before... Wlan0 is not showing.

Mas0x, 2017-05-12:
root@kali:~# python test.py
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.1.46  netmask 255.255.255.0  broadcast 192.168.1.255
        inet6 fe80::ba27:ebff:fe8f:9209  prefixlen 64  scopeid 0x20<link>
        ether b8:27:eb:8f:92:09  txqueuelen 1000  (Ethernet)
        RX packets 12311  bytes 574495 (561.0 KiB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 22761  bytes 31694104 (30.2 MiB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

wlan0: flags=4098<BROADCAST,MULTICAST>  mtu 1400
        inet 10.0.0.1  netmask 255.0.0.0  broadcast 10.255.255.255
        ether 60:e3:27:0a:9d:ac  txqueuelen 1000  (Ethernet)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

restore settings...

mh4x0f, 2017-05-12:
great, I found the problem. before I create um fix for this, I need to test with full code WP. I upload the code fixed in my git dist [main code fix](https://gist.github.com/mh4x0f/636aece16fcd47e0467a6fc39c7a6731). replace the code on core/main.py and try start AP.

Mas0x, 2017-05-12:
Ok just found something interesting...

I updated my Kali to FULL.. apt-get install kali-linux-full

I've not run wifi-pumpkin since, thought I'd try again... not getting that error no more however...

hostapd is giving erros.

mh4x0f, 2017-05-12:
> hostapd is giving erros.

 send me screenshot

Mas0x, 2017-05-12:
Thanks for your time so far on this... not sure what's fixed it but it's up and running!!

But unable to run mana atm. working getting this info for you.

mh4x0f, 2017-05-12:
thk for report :D

Mas0x, 2017-05-12:
# hostapd-mana support 
#enable_mana=1
# Limit mana to responding only  (0), or not (1)
#mana_loud=0
# Extend MAC ACLs to probe frames
#mana_macacl=0
# Put hostapd in white/black list mode
#macaddr_acl=0

Mas0x, 2017-05-12:
Failed to initate Access Point. checkoutput process hostapd.

output::
Configuration file /usr/share/WiFi-Pumpkin/settings/hostapd.conf
Line 5: unknown configuration item 'enable_mana'
Line 6: unknown configuration item 'mana_loud'
Line 7: unknown configuration item 'mana_macacl'
3 errors found in configuration file '/usr/share/WiFi-Pumpkin/settings/hostapd.conf
Failed to set up interface with /usr/share/WiFi-Pumpkin/settings/hostapd.conf
Failed to initialize interface

mh4x0f, 2017-05-12:
if you want to use hostapd-mana, your need to set path the binary the  program hostapd-mana.

Mas0x, 2017-05-12:
I was just looking at the path..

hostapd C. binary

I would think the mana-toolkit has this?

mh4x0f, 2017-05-12:
you need find the path from binary hostapd in directory the mana-toolkit
look this.
[image]

m3010, 2017-05-12:
'_'

Mas0x, 2017-05-12:
git clone https://github.com/sensepost/hostapd-mana.git

I then updated the path to:

/root/hostapd-mana/hostapd/

hostapd binary path is not found.

mh4x0f, 2017-05-12:
> root/hostapd-mana/hostapd/

change to **/root/hostapd-mana/hostapd/hostapd**. after that, compile the hostapd.

Mas0x, 2017-05-12:
Unable to complie 

Makefile:891: recipe for target '../src/crypto/tls_openssl.o' failed

I've read somewhere that the openssl needs to be downgraded in order for it to compile?

I guess I can't use mana until it fixed?

mh4x0f, 2017-05-12:
> hostname WRT54G
> Current MAC: 60:e3:27:0a:9d:ac (TP-LINK TECHNOLOGIES CO.,LTD.)
> Permanent MAC: 60:e3:27:0a:9d:ac (TP-LINK TECHNOLOGIES CO.,LTD.)
> New MAC: b2:22:e3:d3:55:00 (unknown)
> Network driver didn't actually change to the new MAC!!
> Configuration file: /etc/mana-toolkit/hostapd-mana.conf
> Using interface wlan0 with hwaddr 60:e3:27:0a:9d:ac and ssid "FreeWiFi"
> wlan0: interface state UNINITIALIZED->ENABLED
> wlan0: AP-ENABLED

the mana-toolkit is  alrady installed in your system, as can see above. now your need to find the folder when the hostapd-mana is compiled check in **/etc/mana-toolkit/**  or use locate "hostapd" and get the path the binary.

closed 2017-05-12 by mh4x0f

Mas0x, 2017-05-12:
Still unable to compile ... hostapd-2.6 different error.

nilesh0900, 2018-04-08:
root@kali:~# sudo python test.py
python: can't open file 'test.py': [Errno 2] No such file or directory
root@kali:~# python test.py
python: can't open file 'test.py': [Errno 2] No such file or directory
root@kali:~#