P0cL4bs/WiFi-Pumpkin-deprecated #218: NetCreds error

Mas0x, 2017-05-12:
Running plugin SSLStrip+Dns2-Proxy

***Output***

root@kali:~# wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [OTI2MTk=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 5521 (hostapd)]
[New Thread 5523 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::summary           status:On
TCPProxy::imageCap          status:On
TCPProxy::httpCap           status:On
TCPProxy::emails            status:On

-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[60:e3:27:0a:9d:ac] CH 11
libpng warning: iCCP: known incorrect sRGB profile
[New Thread 1756623984 (Dns2proxy::Capture)]
[New Thread 1748235376 (Netcreds::Credentials)]
 [New Thread 1739846768 (Netcreds::Urls)]
Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()
  File "/usr/share/WiFi-Pumpkin/core/servers/proxy/tcp/intercept.py", line 94, in main
    self.plugins[Active].filterPackets(pkt)
  File "/usr/share/WiFi-Pumpkin/plugins/analyzers/image.py", line 38, in filterPackets
    file_name = 'logs/ImagesCap/%s_%s%s' % (self.session,self.random_char(5), file_extension)
  File "/usr/share/WiFi-Pumpkin/plugins/analyzers/image.py", line 43, in random_char
    return ''.join(random.choice(string.ascii_letters) for x in range(y))
  File "/usr/share/WiFi-Pumpkin/plugins/analyzers/image.py", line 43, in <genexpr>
    return ''.join(random.choice(string.ascii_letters) for x in range(y))
NameError: global name 'string' is not defined

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):TL-WN823N
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no
* Operating System and version: Kali Linux

mh4x0f, 2017-05-13:
> Testing shows HTTPS not removed?

this plugin is a POC doesn't work anymore.

closed 2017-05-13 by mh4x0f

Mas0x, 2017-05-13:
POC = HSTS Bypass?

When using mana-toolkit ***HTTPS*** is remove with browser warning, but HSTS now protects.