P0cL4bs/WiFi-Pumpkin-deprecated #223: replaceimages

iRalCraft, 2017-05-18:
## What's the problem (or question)?

Im trying to use the pumpkin-proxy replaceImages. I have set the image to ""/root/Desktop/image.jpg".
However, i dont seem to have any images replaced on the websites i visit when connected to the pumpAP...

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):
ALFA  AWUS051NH
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no, kali is main os
* Operating System and version: kali linux 2

mh4x0f, 2017-05-18:
change path the image to **/home/root/Desktop/image.jpg**.

infiniteral, 2017-05-19:
Thank you very much.
Huge respect for what your doing here!

Best regards,

infiniteral, 2017-05-19:
Im sorry but it still does not seem to work with me. I tried numerous websites like google slides, google images search and other less advanced ones but they still show their normal image... 

Am i doing something wrong here?

closed 2017-05-19 by mh4x0f