P0cL4bs/WiFi-Pumpkin-deprecated #229: struct.error: argument for 's' must be a string

eduardmolina, 2017-06-01:
## What's the problem (or question)?

root@kali:~/WiFi-Pumpkin-0.8.5# wifi-pumpkin
/usr/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 55, in <module>
    networkcontrol = CLI_NetworkManager() # add all interface avaliable for exclude
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 134, in __init__
    self.interfaces = Refactor.get_interfaces()
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 260, in get_interfaces
    interfaces['IPaddress'] = Refactor.get_Ipaddr(interfaces['activated'][0])
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 296, in get_Ipaddr
    return get_if_addr(card)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/__init__.py", line 48, in get_if_addr
    return socket.inet_ntoa(get_if_raw_addr(iff))
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 99, in get_if_raw_addr
    return get_if(iff, SIOCGIFADDR)[20:24]
  File "/usr/lib/python2.7/dist-packages/scapy/arch/common.py", line 19, in get_if
    ifreq = ioctl(sck, cmd, struct.pack("16s16x", iff))
struct.error: argument for 's' must be a string
root@kali:~/WiFi-Pumpkin-0.8.5# 
-------------------------------------------------------------------------------------------------------------------
Pls, help me to solve this

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Qualcomm Atheros QCA9377 802.11ac Wireless Network Adapter
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no
* Operating System and version: Kali Linux 2.0

closed 2017-06-01 by eduardmolina

arpwey, 2017-06-02:
Were you able to fix this issue?  I am getting this error repeatedly.

eduardmolina, 2017-06-02:
No mate, I continue have the same problem....but take a look: https://github.com/P0cL4bs/WiFi-Pumpkin/issues/230

Cooper-Dale, 2017-06-02:
Have same problem on Parrot 3.6; python 2.7.13

mh4x0f, 2017-06-06:
#232 look