P0cL4bs/WiFi-Pumpkin-deprecated #230: WiFi-Pumpkin do not starts after the installation

AcidMand04, 2017-06-02:
## What's the problem (or question)?

*# wifi-pumpkin
/usr/lib/python2.7/dist-packages/matplotlib/font_manager.py:273: UserWarning: Matplotlib is building the font cache using fc-list. This may take a moment.
  warnings.warn('Matplotlib is building the font cache using fc-list. This may take a moment.')
Loading GUI...
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 53, in <module>
    app = Initialize()
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 94, in __init__
    self.form_widget    = WifiPumpkin(self,self,self.FSettings)
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 160, in __init__
    self.UpdateSoftware = frm_githubUpdate(version)
  File "/usr/share/WiFi-Pumpkin/core/helpers/update.py", line 32, in __init__
    super(frm_githubUpdate, self).__init__(parent)
  File "/usr/share/WiFi-Pumpkin/core/loaders/models/PackagesUI.py", line 40, in __init__
    self.interfaces     = Refactor.get_interfaces()
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 260, in get_interfaces
    interfaces['IPaddress'] = Refactor.get_Ipaddr(interfaces['activated'][0])
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 296, in get_Ipaddr
    return get_if_addr(card)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/__init__.py", line 48, in get_if_addr
    return socket.inet_ntoa(get_if_raw_addr(iff))
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 99, in get_if_raw_addr
    return get_if(iff, SIOCGIFADDR)[20:24]
  File "/usr/lib/python2.7/dist-packages/scapy/arch/common.py", line 19, in get_if
    ifreq = ioctl(sck, cmd, struct.pack("16s16x", iff))
struct.error: argument for 's' must be a string
*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Atheros UB91C
* Version used tool: 
* Virtual Machine (yes or no and which):  VMware Latest
* Operating System and version:  Kali 2, Latest

Crossik9, 2017-06-03:
`pip install mitmproxy==0.17`

AcidMand04, 2017-06-03:
`pip install mitmproxy==0.17`
I've receive an answer, which I suppose had to fix this issue ?
But unfortunately it is Not. it ends as follow: 

`    ----------------------------------------
  Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-fL5rmG/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-FBTcx5-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-fL5rmG/cryptography/`

then trying to run WiFi-Pumpkin again, and getting the following:

`wifi-pumpkin
Loading GUI...
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 53, in <module>
    app = Initialize()
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 94, in __init__
    self.form_widget    = WifiPumpkin(self,self,self.FSettings)
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 160, in __init__
    self.UpdateSoftware = frm_githubUpdate(version)
  File "/usr/share/WiFi-Pumpkin/core/helpers/update.py", line 32, in __init__
    super(frm_githubUpdate, self).__init__(parent)
  File "/usr/share/WiFi-Pumpkin/core/loaders/models/PackagesUI.py", line 40, in __init__
    self.interfaces     = Refactor.get_interfaces()
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 260, in get_interfaces
    interfaces['IPaddress'] = Refactor.get_Ipaddr(interfaces['activated'][0])
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 296, in get_Ipaddr
    return get_if_addr(card)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/__init__.py", line 48, in get_if_addr
    return socket.inet_ntoa(get_if_raw_addr(iff))
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 99, in get_if_raw_addr
    return get_if(iff, SIOCGIFADDR)[20:24]
  File "/usr/lib/python2.7/dist-packages/scapy/arch/common.py", line 19, in get_if
    ifreq = ioctl(sck, cmd, struct.pack("16s16x", iff))
struct.error: argument for 's' must be a string
`

mh4x0f, 2017-06-06:
#232 look

closed 2017-06-06 by mh4x0f