P0cL4bs/WiFi-Pumpkin-deprecated #232: Issues upon start

nubbix, 2017-06-03:
Using Ubuntu latest version 
wlan1 - alpha

Getting this issue

Loading GUI...
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 53, in <module>
    app = Initialize()
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 94, in __init__
    self.form_widget    = WifiPumpkin(self,self,self.FSettings)
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 160, in __init__
    self.UpdateSoftware = frm_githubUpdate(version)
  File "/usr/share/WiFi-Pumpkin/core/helpers/update.py", line 32, in __init__
    super(frm_githubUpdate, self).__init__(parent)
  File "/usr/share/WiFi-Pumpkin/core/loaders/models/PackagesUI.py", line 40, in __init__
    self.interfaces     = Refactor.get_interfaces()
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 260, in get_interfaces
    interfaces['IPaddress'] = Refactor.get_Ipaddr(interfaces['activated'][0])
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 296, in get_Ipaddr
    return get_if_addr(card)
  File "/usr/lib/python2.7/dist-packages/scapy/arch/__init__.py", line 36, in get_if_addr
    return socket.inet_ntoa(get_if_raw_addr(iff))
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 86, in get_if_raw_addr
    return get_if(iff, SIOCGIFADDR)[20:24]
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 269, in get_if
    ifreq = ioctl(s, cmd, struct.pack("16s16x",iff))
struct.error: argument for 's' must be a string

ylot0, 2017-06-04:
ubuntu 16.04LTS

struct.error: argument for 's' must be a string

okazymyrov, 2017-06-05:
An intermediate fix... In "/usr/share/WiFi-Pumpkin/core/utils.py" (function get_Ipaddr) change the second return to:
`return get_if_addr("{0}".format(card))`

mh4x0f, 2017-06-05:
@okazymyrov thanks for suggestion.

AcidMand04, 2017-06-06:
did u guys update the main distro ?

eisyeisy, 2017-06-06:
how I can open and rewrite the utils.py?

eisyeisy, 2017-06-06:
After I rewrite the function get_Ipaddr, I become the following issue:

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 39, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.utils import Refactor
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 296
    return get_if_addr("{0}".format(card))
                                         ^
IndentationError: unindent does not match any outer indentation level

okazymyrov, 2017-06-06:
@eisyeisy you need to ensure that indentation level is the same after editing. Most likely your editor has changed indentation to "tab" while original level was 4 spaces.

eisyeisy, 2017-06-06:
@okazymyrov thank you!
but now I became this issue:

Traceback (most recent call last):
  File "wifi-pumpkin.py", line 39, in <module>
    from core.loaders.checker.networkmanager import CLI_NetworkManager,UI_NetworkManager
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/networkmanager.py", line 3, in <module>
    from core.utils import Refactor
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 9, in <module>
    from scapy.all import *
  File "/usr/lib/python2.7/dist-packages/scapy/all.py", line 28, in <module>
    from scapy.route6 import *
  File "/usr/lib/python2.7/dist-packages/scapy/route6.py", line 273, in <module>
    conf.route6 = Route6()
  File "/usr/lib/python2.7/dist-packages/scapy/route6.py", line 31, in __init__
    self.resync()
  File "/usr/lib/python2.7/dist-packages/scapy/route6.py", line 44, in resync
    self.routes = read_routes6()
  File "/usr/lib/python2.7/dist-packages/scapy/arch/linux.py", line 283, in read_routes6
    cset = scapy.utils6.construct_source_candidate_set(d, dp, devaddrs, LOOPBACK_NAME)
  File "/usr/lib/python2.7/dist-packages/scapy/utils6.py", line 50, in construct_source_candidate_set
    if in6_isgladdr(addr) or in6_isuladdr(addr):
  File "/usr/lib/python2.7/dist-packages/scapy/utils6.py", line 708, in in6_isgladdr
    return in6_isincluded(str, '2000::', 3)
  File "/usr/lib/python2.7/dist-packages/scapy/utils6.py", line 651, in in6_isincluded
    temp = inet_pton(socket.AF_INET6, addr)
  File "/usr/lib/python2.7/dist-packages/scapy/pton_ntop.py", line 40, in inet_pton
    raise Exception("Illegal syntax for IP address")
Exception: Illegal syntax for IP address

closed 2017-06-06 by mh4x0f

gitxmax, 2018-01-24:
Hello,
I have a fresh install of Pumpkin.
When I launch the "ARP POISONER" module, i get this error 
`Traceback (most recent call last):
  File "/usr/share/WiFi-Pumpkin/core/main.py", line 680, in show_arp_posion
    self.Farp_posion = GUIModules.frm_Arp_Poison(self.window_phishing)
  File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 42, in __init__
    self.GUI()
  File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 145, in GUI
    self.ConfigureEdits()
  File "/usr/share/WiFi-Pumpkin/modules/poisoners/arp_poisoner.py", line 174, in ConfigureEdits
    self.txt_mac.setText(Refactor.getHwAddr(x['activated'][0]))
  File "/usr/share/WiFi-Pumpkin/core/utils.py", line 237, in getHwAddr
    info = ioctl(s.fileno(), 0x8927,  pack('256s', ifname[:15]))
struct.error: argument for 's' must be a string`

I tried the solution offered here to modify it to: return get_if_addr("{0}".format(card))

 This had no effect, same error persist.

Running Kali 17.1 on raspberry 3