P0cL4bs/WiFi-Pumpkin-deprecated #238: Works with ethernet but cannot manage to make it work with 2 wireless

debosset, 2017-06-13:
## What's the problem (or question)?

I've been using my Alfa AWUS036H with WiFi-Pumpkin with internet connection ok
But then i want to use the broadcom wifi on the card so my alfa can go in monitor mode that's fine and i'm using a simple one that can just connect to my internet (to replace my rj45).

This is what i do. I start kali on my raspberry, then when i'm logged in i connect to my wifi with the wlan0 (which is the simple broadcom)

Then i run sudo wifi-pumpkin and when the gui is loaded. I'm choosing in the adapter field wlan1

Then it just doesn't work. What i'm doing wrong?

Thanks in advance for your help

*answer here*

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): Alfa AWUS036H
* Version used tool: WiFi-Pumpkin v0.8.5
* Virtual Machine (yes or no and which): no , Kali Rolling on raspberry pi (updated)
* Operating System and version: Kali Rolling but it has been updated

usama7628674, 2017-06-13:
Give detail about error.Does it showing nl80211 error?

debosset, 2017-06-14:
Hi,

No i'm not getting nl80211 error. I'm getting : 

Connection has been detected, this wlan0 is joined the correct Wi-Fi network : Device or ressource busy
wlan0 : IEEE 802.11bgn
ESSID : "Mywifi"

You may need to another Wi-Fi USB adapter... etc...

What i'm doing wrong?

Thanks for your help

nemew, 2017-06-14:
Are you selecting your free wifi adapter in the settings tab? sounds like you are trying to use the one connected to the internet. in my case i select wlan1 and it works just fine

debosset, 2017-06-14:
If i select wlan, it says that the adapter doesn't suport the monitor mode. But it's weird since the Alfa AWUS036H is working when internet is comming feom my rj45. I have been reinstalling kali re4son on my raspberry and install full kali. But the problem remain the same. Ia there a settings where i can choose which wlan i want to get internet?
I found where to define for the ap. Thanks in advance for your help.

So i have to connect through a terminal or just clicking on the right corner and connect is fine? And then i run wifi-pumpkin.

I don't know what i'm doing wrong. Hope we find a solution :) thanks in advance for your help

debosset, 2017-06-14:
I have wlan0 (broadcom) and wlan1 (Alfa AWUS036H)

nemew, 2017-06-16:
You have to only have one of your wifi adapter connected to the internet. (the one you will not be using as an AP)
then when you start wifi-pumpkin it will automatically detect which one is connected. Once you are in the settings tab. Select the adapter which is NOT connected to the internet and hit start. It should work

nemew, 2017-06-16:
in your case. connect to the internet with wlan0. And in the settings tabs use wlan1 from the list

closed 2017-07-01 by mh4x0f

gasserahmad, 2019-12-28:
step 1 : disconnect internet from your virtual machine 
step 2 : you should have 2 (tow) interfaces  connected to your device 1 to connect to your wifi that’s will be wlan0 and 2 to the fake wifi point and that’s will be wlan1 
step 3 : from the WiFi-Pumpkin setting select wlan1 and press Start 
and should be work fine ✅