P0cL4bs/WiFi-Pumpkin-deprecated #239: USB wifi not working properly after using AP

nemew, 2017-06-14:
After using my usb wifi adapter a few times the program crashed while enabling the dns spoof. After that I can still see my adapter "wlan1" in the list to enable the AP. But it shows up as blank in my wifi drop-down menu at the top of my kali screen. 
when I turn off my wifi adapters the name appears as "USB-WIFI" but when i turn them back on only the wifi symbol stays and the name dissapears. I am unable to use it to connect to networks through network-manager. 

#### Please tell us details about your environment.
* Card wireless adapter which has the problem - alfa awus063nha
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): NO
* Operating System and version: Kali Linux 2017.2

usama7628674, 2017-06-15:
Here's you can do to resolve the problem.
First open terminal and type
nano /etc/NetworkManager/network-manager.conf
look for unmanaged-devices and change your mac address to alfa's adapter. 
save and close the terminal
open again terminal and type
service network-manager restart
then use Wifi Pumpkin
After using it go to same network-manager.conf and erase the following two lines
[Keyfile]
unmanaged-devices:XX:XX:XX:XX:XX:XX

X is your mac address of usb adapter.

nemew, 2017-06-16:
Thanks for the reply. when i use "nano /etc/NetworkManager/network-manager.conf" it shows up blank.
I checked the directory and there is only  a file called  "NetworkManager.conf".  I opened it and this is what it has in it.

[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

I have this problem on my local Kali install and now after I stopped using wifi-pumpkin with the dns spoofer and phishing manager. I have the same problem on my VM kali install.

nemew, 2017-06-17:
UPDATE:
The solution provided solved the issue on my local kali installation. Thank you!
The VM kali install was solved by just restarting the network-manager

nemew, 2017-06-17:
UPDATE2
Ok so this made my wifi adapter work again. But it stopped wifi-pumpkin from creating the AP. I was getting the hostpad error which is everywhere in the issues.

I tried everything. And the ONLY way it ended up working was to have wifi-pumpkin open. Turn on airmon-ng on the interface you will use as the ap. Refresh the interface list in wifi-pumpkin. select wlan1mon and start the ap. WORKS!! weird. maybe a bug?

usama7628674, 2017-06-19:
Actually you have to unmanage your USB adapter to avoid hostpad error.To do that just do the folllowing:-
1. Go to NetworkManager.conf using nano editor in terminal.
 [Keyfile]
 unmanaged-devices:XX:XX:XX:XX:XX:XX
2. Write your  MAC address of  USB adapter in place of X.
3. Save it.
4. Restart network manager using terminal 
    service network-manager restart.
5. Then check network manager at top right and you will see your USB adapter in unmanaged mode and only internal adpater is connected to internet.
6. Now wifi-pumpkin should work.

usama7628674, 2017-06-19:
Your NetworkManager.conf should look like this
[main]
plugins=ifupdown,keyfile

[ifupdown]
managed=false

[Keyfile]
unmanaged-devices:XX:XX:XX:XX:XX:XX (Write your wifi adapter MAC address here which you would use to create AP in Wifi-pumpkin)

commit 9c0f9346 (2017-07-13): added option for restore NM USB adpater after app closed #239
closed 2017-07-13 by mh4x0f