P0cL4bs/WiFi-Pumpkin-deprecated #240: sudo: wifi-pumpkin: command not found !!!! although It's supposed to be installed

x-cod3r, 2017-06-15:
""sudo: wifi-pumpkin: command not found""

this is the answer after installing wifi-pumpkin and trying to run it,,simply all diles disappear from usr/share/WiFi-Pumpkin and when trying to run the command from other place it give me not found answer however it's supposed to be installed on my machine 

one last thing, this is the reply after running ./install-sh install command .......it downloaded some repositories and this was the last part from the reply....

##################################################################
[=] checking dependencies  
----[✔]----[+] hostapd Installed
----------------------------------------
[+] Distribution Name: Parrot
----------------------------------------
[=]  Install WiFi-Pumpkin 
cp: cannot stat '/usr/share/WiFi-Pumpkin/*': No such file or directory
[✔] binary::/usr/bin/
sh: 0: getcwd() failed: No such file or directory
update-alternatives: error: alternative path /usr/share/WiFi-Pumpkin/wifi-pumpkin doesn't exist
cp: cannot stat 'wifi-pumpkin.desktop': No such file or directory
chmod: cannot access '/usr/share/WiFi-Pumpkin/wifi-pumpkin': No such file or directory
[✔] wifi-pumpkin installed with success
[✔] execute  sudo wifi-pumpkin in terminal
[+] P0cL4bs Team CopyRight 2015-2017
[+] Enjoy
######################################################################

I'm using parrot os persistent USB disck
and this is the system info >>>
***********************************
DISTRIB_ID=Parrot
DISTRIB_RELEASE=3.6
DISTRIB_CODENAME=JollyRoger
DISTRIB_DESCRIPTION="Parrot 3.6 - JollyRoger"
PRETTY_NAME="Parrot GNU/Linux 3.6 JollyRoger"
NAME="Parrot GNU/Linux"
ID=parrot
VERSION="3.6 (JollyRoger)"
VERSION_ID="3.6"
ID_LIKE=debian
ANSI_COLOR="1;31"
HOME_URL="https://www.parrotsec.org/"
SUPPORT_URL="https://community.parrotsec.org/"
BUG_REPORT_URL="https://dev.parrotsec.org/"
*******************

nemew, 2017-06-16:
are you typing "sudo: wifi-pumpkin" or "sudo wifi-pumpkin"?
if your using "sudo:" remove the ":"

x-cod3r, 2017-06-17:
Hi @nemew 
I'm writing "sudo wifi-pumpkin" 
This ":" only added in the response
It's a real problem here.....after installation all the files simply disapppear from the machine to somewhere which I can't locate......usr/share/Wifi-Pumpkin get empty after every try I did

x-cod3r, 2017-06-17:
@mh4x0f  @TimSchumi  plz help!

closed 2017-07-09 by mh4x0f