P0cL4bs/WiFi-Pumpkin-deprecated #243: ALFA card (AWUS036NHA) dropping connection (connects and disconnects) after some seconds

usama7628674, 2017-06-18:
## 
     this issue happens every time when I use WiFi Pumpkin for the first time in kali linux or parror os
     and then I couldn't fix it.I also restarted linux but all in vain.I checked NetworkManager.conf file and
      manage and unmanage the device but nothing happens.I'm using alfa as a normal internet                          
      connection.
       Any help would be greatly appreciated.

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):ALFA AWUS036NHA
* Version used tool: Wifi pumpkin 0.8.5
* Virtual Machine (yes or no and which):no 
* Operating System and version: parrot os 3.6

closed 2017-07-02 by mh4x0f

Robinson-George, 2020-01-24:
I have experienced the same problems with this adapter and after debugging the drivers for a long time, I did not find the problem.  Then I have noticed that the same problems also exists under Windows, which were installed on the same machine (with dual boot). This would make driver problems a hell of a coincidence so I started suspecting, that the problem is with this WiFi adapter's hardware or with the USB port that serves it. Therefore, I returned the old and reordered a new `ALFA AWUS036NHA` WiFi adapter and installed a new PCIe card with a pair of new USB ports.  

To my surprise the problems persisted ON BOTH operating systems!

Remembering, that the LED indicator light of this WiFi adapter sometimes freezes ( mostly on the OFF state ), I started suspecting that the MCU inside this WiFi adapter becomes hung or reset due to power supply issues, so I plugged an `inline USB power monitor` into the Mini-B USB socket of the WiFi adapter, set it to `MAX Hold display` and discovered, that this adapter draws **more** current than the maximum allowed by the USB standard ( which is 500mA ) by intermittently drawing up to 830mA on stock settings (and up to 910mA with the [increased TX power hack][1] ).  

After constructing my own 3meter long USB cable out of a 90Ω impedance shielded twisted pair cable ( S/STP with 16AWG conductors ) with two paralleled USB Type-A jacks at one end of this cable, for drawing the supply current from 2 host USB sockets at the same time (but the data only form one USB socket!), all of the random disconnection problems, hangs and lockup probes have disappeared.

  [1]: https://miloserdov.org/?p=337

usama7628674, 2020-01-24:
@Robinson-George The simple regulatory hacks and driver patches wont do anything at all according to this https://github.com/qca/open-ath9k-htc-firmware/issues/82#issuecomment-519023480
and https://github.com/qca/open-ath9k-htc-firmware/issues/82#issuecomment-231666149

Are you sure the adapter draws 910mA after patching regulatory domain?

Robinson-George, 2020-02-03:
Yes, but note that I only measured supply input current, not RF output power.