P0cL4bs/WiFi-Pumpkin-deprecated #247: No traffic :()

GGGunrunner, 2017-06-26:
Hello everyone!

Trying to get it working. Kali Virtualbox, eth0 is connected, wlan0 is TL-WN722N.

Pumpkin starts without any errors. Client (Android) connects. But no traffic is going in or out.
I checked all Youtube videos, they show exactly the same setup and it works for them.

At first I thought there must be something wrong with iptables or forwarding. No, everything seems ok.

Can anybody suggest?

--

WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [MTcxOTE=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 12107 (hostapd)]
[*] Pumpkin-Proxy running on port:8080 

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[ca:1b:2d:19:1c:d5] CH 11
-------------------------------
Thread::[hostapd] successfully stopped.
Thread::[Pumpkin-Proxy] successfully stopped.
Thread::[TCPProxy] successfully stopped.

[*] Loading debugging mode
[*] Current Session::ID [OTE4MDY=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 12204 (hostapd)]
[*] Pumpkin-Proxy running on port:8080 

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::httpCap           status:On
TCPProxy::imageCap          status:On
TCPProxy::summary           status:On
TCPProxy::emails            status:On

-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[3a:1b:3d:19:3c:f5] CH 11

mh4x0f, 2017-07-03:
#43

mh4x0f, 2017-07-03:
I don't find any solution for now. but try change the mode bridge to nat  or nat to bridge and check if working

closed 2017-07-09 by mh4x0f