P0cL4bs/WiFi-Pumpkin-deprecated #248: Credentials in terminal. Nothing in GUI

argtje99, 2017-07-01:
Hello,
I have the following issue.
When i start Wifi-Pumpkin everything works fine. except a small thing. I dont see any traffic in the NetCreds. It just monitor for 2 min and from then nothing more.

[image]

As you see at the first time it was succesful. But the second time you see the credentials in terminal. But not in the HTTP-Authentication or the NetCreds Logger

Maybe you can help me 
Thankyou

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode): TPLINK WN722n
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): No
* Operating System and version: Kali Linux Rolling 2017.1

mh4x0f, 2017-07-01:
solution in progress look #234

argtje99, 2017-07-01:
Thanks @mh4x0f

argtje99, 2017-07-02:
@mh4x0f i get this log
---------------------------------------------------
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [NzY2NTg=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 11679 (hostapd)]
[New Thread 11681 (Dns2Proxy)]
SSLstrip v0.9 + POC by Leonardo Nve

[*] TCPProxy running on port 80/8080:

TCPProxy::httpCap           status:On

-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[f4:ec:38:8f:e6:bb] CH 4
Traceback (most recent call last):
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/servers/proxy/tcp/intercept.py", line 94, in main
    self.plugins[Active].filterPackets(pkt)
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/analyzers/httpCap.py", line 87, in filterPackets
    self.load = pkt[Raw].load
  File "/usr/lib/python2.7/dist-packages/scapy/packet.py", line 817, in __getitem__
    raise IndexError("Layer [%s] not found" % lname)
IndexError: Layer [Raw] not found

argtje99, 2017-07-02:
Unhandled Error
Traceback (most recent call last):
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/utility/threads.py", line 253, in run
    reactor.run(installSignalHandlers=False)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1199, in run
    self.mainLoop()
  File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1211, in mainLoop
    self.doIteration(t)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 235, in doPoll
    log.callWithLogger(selectable, _drdw, selectable, fd, event)
--- <exception caught here> ---
  File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 103, in callWithLogger
    return callWithContext({"system": lp}, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 86, in callWithContext
    return context.call({ILogContext: newCtx}, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
    return self.currentContext().callWithContext(ctx, func, *args, **kw)
  File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 81, in callWithContext
    return func(*args,**kw)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 610, in _doReadOrWrite
    self._disconnectSelectable(selectable, why, inRead)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 252, in _disconnectSelectable
    selectable.readConnectionLost(f)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 272, in readConnectionLost
    self.connectionLost(reason)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 478, in connectionLost
    self._commonConnection.connectionLost(self, reason)
  File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 292, in connectionLost
    protocol.connectionLost(reason)
  File "/usr/lib/python2.7/dist-packages/twisted/protocols/tls.py", line 503, in connectionLost
    ProtocolWrapper.connectionLost(self, reason)
  File "/usr/lib/python2.7/dist-packages/twisted/protocols/policies.py", line 125, in connectionLost
    self.wrappedProtocol.connectionLost(reason)
  File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 486, in connectionLost
    self.handleResponseEnd()
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/external/sergio_proxy/sslstrip/ServerConnection.py", line 131, in handleResponseEnd
    HTTPClient.handleResponseEnd(self)
  File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 497, in handleResponseEnd
    self.handleResponse(b)
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/external/sergio_proxy/sslstrip/ServerConnection.py", line 163, in handleResponse
    self.client.write(data)
  File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 896, in write
    raise RuntimeError('Request.write called on a request after '
exceptions.RuntimeError: Request.write called on a request after Request.finish was called.

usama7628674, 2017-07-02:
look here #234
I already opened this issue when you get "layer [RAW] not found" when using pumpkin proxy but you get
different runtime errors when using sslstrip+dnsproxy or sergio-proxy.Hopefully it will be resolved soon

usama7628674, 2017-07-02:
WiFi-Pumpkin assumes you should get a Raw layer somewhere, although in practice you don't. It seems the problem comes from the WiFi-Pumpkin code itself.So clearly not a bug in scapy.

argtje99, 2017-07-02:
I see @usama7628674 

_Sent from my Samsung SM-G5500 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

closed 2017-07-03 by mh4x0f

argtje99, 2017-07-03:
@mh4x0f No Layer Raw error now. Check this.

Traceback (most recent call last):
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/servers/proxy/tcp/intercept.py", line 43, in run
self.main()
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/servers/proxy/tcp/intercept.py", line 94, in main
self.plugins[Active].filterPackets(pkt)
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/analyzers/image.py", line 40, in filterPackets
urlretrieve('http://{}{}'.format(http_layer.fields['Host'], http_layer.fields['Path']),file_name)
File "/usr/lib/python2.7/urllib.py", line 98, in urlretrieve
return opener.retrieve(url, filename, reporthook, data)
File "/usr/lib/python2.7/urllib.py", line 245, in retrieve
fp = self.open(url, data)
File "/usr/lib/python2.7/urllib.py", line 213, in open
return getattr(self, name)(url)
File "/usr/lib/python2.7/urllib.py", line 350, in open_http
h.endheaders(data)
File "/usr/lib/python2.7/httplib.py", line 1038, in endheaders
self._send_output(message_body)
File "/usr/lib/python2.7/httplib.py", line 882, in _send_output
self.send(msg)
File "/usr/lib/python2.7/httplib.py", line 844, in send
self.connect()
File "/usr/lib/python2.7/httplib.py", line 821, in connect
self.timeout, self.source_address)
File "/usr/lib/python2.7/socket.py", line 557, in create_connection
for res in getaddrinfo(host, port, 0, SOCK_STREAM):
IOError: [Errno socket error] [Errno -2] Name or service not known

After that i get this error

Unhandled Error
Traceback (most recent call last):
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/utility/threads.py", line 253, in run
reactor.run(installSignalHandlers=False)
File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1199, in run
self.mainLoop()
File "/usr/lib/python2.7/dist-packages/twisted/internet/base.py", line 1211, in mainLoop
self.doIteration(t)
File "/usr/lib/python2.7/dist-packages/twisted/internet/epollreactor.py", line 235, in doPoll
log.callWithLogger(selectable, _drdw, selectable, fd, event)
--- ---
File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 103, in callWithLogger
return callWithContext({"system": lp}, func, *args, **kw)
File "/usr/lib/python2.7/dist-packages/twisted/python/log.py", line 86, in callWithContext
return context.call({ILogContext: newCtx}, func, *args, **kw)
File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 118, in callWithContext
return self.currentContext().callWithContext(ctx, func, *args, **kw)
File "/usr/lib/python2.7/dist-packages/twisted/python/context.py", line 81, in callWithContext
return func(*args,**kw)
File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 610, in _doReadOrWrite
self._disconnectSelectable(selectable, why, inRead)
File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 255, in _disconnectSelectable
selectable.connectionLost(f)
File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 478, in connectionLost
self._commonConnection.connectionLost(self, reason)
File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 292, in connectionLost
protocol.connectionLost(reason)
File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 486, in connectionLost
self.handleResponseEnd()
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/external/sslstrip/ServerConnection.py", line 133, in handleResponseEnd
HTTPClient.handleResponseEnd(self)
File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 497, in handleResponseEnd
self.handleResponse(b)
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/external/sslstrip/ServerConnection.py", line 163, in handleResponse
self.client.write(data)
File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 896, in write
raise RuntimeError('Request.write called on a request after '
exceptions.RuntimeError: Request.write called on a request after Request.finish was called.

[... 19 lines cut ...]
File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 610, in _doReadOrWrite
self._disconnectSelectable(selectable, why, inRead)
File "/usr/lib/python2.7/dist-packages/twisted/internet/posixbase.py", line 255, in _disconnectSelectable
selectable.connectionLost(f)
File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 478, in connectionLost
self._commonConnection.connectionLost(self, reason)
File "/usr/lib/python2.7/dist-packages/twisted/internet/tcp.py", line 292, in connectionLost
protocol.connectionLost(reason)
File "/usr/lib/python2.7/dist-packages/twisted/protocols/tls.py", line 503, in connectionLost
ProtocolWrapper.connectionLost(self, reason)
File "/usr/lib/python2.7/dist-packages/twisted/protocols/policies.py", line 125, in connectionLost
self.wrappedProtocol.connectionLost(reason)
File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 486, in connectionLost
self.handleResponseEnd()
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/external/sslstrip/ServerConnection.py", line 133, in handleResponseEnd
HTTPClient.handleResponseEnd(self)
File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 497, in handleResponseEnd
self.handleResponse(b)
File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/external/sslstrip/ServerConnection.py", line 163, in handleResponse
self.client.write(data)
File "/usr/lib/python2.7/dist-packages/twisted/web/http.py", line 896, in write
raise RuntimeError('Request.write called on a request after '
exceptions.RuntimeError: Request.write called on a request after Request.finish was called.

_Sent from my Samsung SM-G5500 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_