P0cL4bs/WiFi-Pumpkin-deprecated #249: IOError after searching HTTPS websites on target

argtje99, 2017-07-03:
LOG 
-----------------------------------------------------------------

Traceback (most recent call last):
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/servers/proxy/tcp/intercept.py", line 43, in run
    self.main()
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/core/servers/proxy/tcp/intercept.py", line 94, in main
    self.plugins[Active].filterPackets(pkt)
  File "/home/hargerao/Documents/Tools/WiFi-Pumpkin-master/plugins/analyzers/image.py", line 40, in filterPackets
    urlretrieve('http://{}{}'.format(http_layer.fields['Host'], http_layer.fields['Path']),file_name)
  File "/usr/lib/python2.7/urllib.py", line 98, in urlretrieve
    return opener.retrieve(url, filename, reporthook, data)
  File "/usr/lib/python2.7/urllib.py", line 245, in retrieve
    fp = self.open(url, data)
  File "/usr/lib/python2.7/urllib.py", line 213, in open
    return getattr(self, name)(url)
  File "/usr/lib/python2.7/urllib.py", line 350, in open_http
    h.endheaders(data)
  File "/usr/lib/python2.7/httplib.py", line 1038, in endheaders
    self._send_output(message_body)
  File "/usr/lib/python2.7/httplib.py", line 882, in _send_output
    self.send(msg)
  File "/usr/lib/python2.7/httplib.py", line 844, in send
    self.connect()
  File "/usr/lib/python2.7/httplib.py", line 821, in connect
    self.timeout, self.source_address)
  File "/usr/lib/python2.7/socket.py", line 557, in create_connection
    for res in getaddrinfo(host, port, 0, SOCK_STREAM):
IOError: [Errno socket error] [Errno -2] Name or service not known

argtje99, 2017-07-03:
@mh4x0f

usama7628674, 2017-07-06:
It's fine.I checked https sites are working fine no IO error.I'm using pumpkin proxy along with TCP proxy

closed 2017-07-09 by mh4x0f

argtje99, 2017-07-12:
They are working fine. @usama7628674 but no HSTS on the website. Like the sslstrip or dns2proxy isnt working properly