P0cL4bs/WiFi-Pumpkin-deprecated #250: Usb adapter disconnects after few seconds

usama7628674, 2017-07-04:
After configuring when I click on start button in GUI it works for few seconds and then USB WiFi-adapter disconnects for say 2-3 seconds and connects back again and this continues even when I'm not using Wifi-pumpkin and so Fake AP vanishes from devices that are using it and I have to start again and this whole process continues.
I'm using wlan0 as my internal wifi device which is connected to internet and wlan1 for creating rogue 
AP.

root@kali:~# wifi-pumpkin
Loading GUI...
WiFi-Pumpkin Running!

[*] Loading debugging mode
[*] Current Session::ID [OTMzNjg=]
[*] Configuring hostapd...
[*] enable forwarding in iptables...
[*] Configuring dhcpd...
[*] Sharing Internet Connections with NAT...
[New Thread 15420 (hostapd)]
[*] Pumpkin-Proxy running on port:8080 

[*] TCPProxy running on port 80/8080:

TCPProxy::ftp               status:On
TCPProxy::hexdump           status:On
TCPProxy::kerberos          status:On
TCPProxy::summary           status:On
TCPProxy::imageCap          status:On
TCPProxy::httpCap           status:On
TCPProxy::emails            status:On

-------------------------------
AP::[PumpAP] Running...
AP::BSSID::[00:c0:ca:84:ad:95] CH 11
tcpdump: wlan1: No such device exists
(SIOCGIFHWADDR: No such device)

#### Please tell us details about your environment.
* Card wireless adapters name (please check if support AP/mode):alfa awus036nha
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which): no
* Operating System and version: kali 2017.1

argtje99, 2017-07-05:
Did you put your wlan1 into 'unmanaged mode'? if not try that first @usama7628674

usama7628674, 2017-07-06:
I add Keyfile and un-managed my wlan1 adapter from NetworkManager.conf file.But problem still remains.
When I freshly installed kali wlan1 adapter works fine but after using wifi-pumpkin just for one time
the problem starts to occur and it becomes permanent

usama7628674, 2017-07-06:
I think it time out for few seconds.

mh4x0f, 2017-07-09:
look [tcpdump:bug](http://seclists.org/tcpdump/2009/q1/66), this bug come from:
tcpdump version 3.9.7
libpcap version 0.9.7. 
I recommend to try install 0.8 is more stable.

usama7628674, 2017-07-10:
Sorry I don't get you.Are you saying that I need to install libpcap version 0.8 or tcpdump version 0.8
or I need to install both tools their old version because right now I have installed their new version.
Here's what dpkg -l says

libpcap-dev    1.8.1-3
libpcap0.8:amd 1.8.1-3
libpcap0.8-dev 1.8.1-3
tcpdump        4.9.0-2

closed 2017-07-13 by mh4x0f