P0cL4bs/WiFi-Pumpkin-deprecated #255: Bugs

Akhipbm, 2017-07-17:
Respected sir 

when i start the program it show me the below error. i have put the screens shot please check 

im using tenda w311m wireless adapter card

linux is parrot security OS.

I'm using virtual machine VMware 10 workstation.

when i click on start attack .it show me that this message is received because it has bugs. please check the screen shot. its written under the start attack

*

- 

- 

- 

- AP/mode):
[image]

usama7628674, 2017-07-18:
The version you are using is old.Use new version or clone latest repo and then try again

Akhipbm, 2017-07-18:
Please send me the link of the new one.

usama7628674, 2017-07-18:
type this in terminal of your linux
git clone https://github.com/P0cL4bs/WiFi-Pumpkin.git
then this

cd WiFi-Pumpkin
and after that 

./installer.sh --install

usama7628674, 2017-07-18:
and it will install and after installation type this in terminal
sudo wifi-pumpkin or wifi-pumpkin

Akhipbm, 2017-07-18:
Respected Sir

I tried the above thing but when i click on start it showed me the
following error. please check the screen shot.
i even tried to uncheck the option under settings - ignore usb wifi adapter
permanently still its not working.

Config

Parrost OS 3
Vm ware 10
 Card wireless adapters name : tenda usb w311m

usama7628674, 2017-07-18:
please describe error here.I couldn't see properly from image above

Akhipbm, 2017-07-18:
It says failed AP mode warning interface. Your adaptor does support create
acces point

Akhipbm, 2017-07-18:
It also shows .The feature access point not supported by the device wlan0.
But my card support s AP.

usama7628674, 2017-07-18:
Unfortunately your adapter doesn't support rogue AP.Consider buying new adapter like alfa awus036nha or 
tp link wn722

usama7628674, 2017-07-18:
or you didn't install drivers properly

usama7628674, 2017-07-18:
try this 
iw dev wlan0 scan replace wlan0 with your interface

usama7628674, 2017-07-18:
see if it supports monitor mode and packet injection

Akhipbm, 2017-07-18:
Ok thank you for helping me out. Thank you for Ur response.

closed 2017-08-09 by mh4x0f