P0cL4bs/WiFi-Pumpkin-deprecated #260: ERROR HOSTPAD

IsaacMaciel, 2017-07-27:
## What's the problem (or question)?

*answer here*

#### Please tell us details about your environment.

Always fails to start the AP, I believe it is because my system language is PT-BR.
I downloaded the IWC and it did nothing.

* Card wireless adapters name (please check if support AP/mode): AWSUS036H
* Version used tool: 0.8.5
* Virtual Machine (yes or no and which):  YES
* Operating System and version:  Kali Linux 2017.1

wikijm, 2017-07-27:
Careful, Alfa AWSUS036H do not support AP mode with hostapd.
You have to use aibase-ng instead, which is not supported by Wifi-Pumpkin.

Regarding to your screenshot I assume that your problem is related to unsupported PT-BR characters too.

IsaacMaciel, 2017-07-27:
Guys I solved my problem. I changed the installation location of my pumpkin and managed to run smoothly.

IsaacMaciel, 2017-07-27:
On my AWUS036H I succeeded in raising an AP, however an AP is slow

closed 2017-08-09 by mh4x0f