P0cL4bs/WiFi-Pumpkin-deprecated #261: Install Issue

alexisaeff, 2017-07-27:
build/temp.linux-x86_64-2.7/_openssl.c:70047:29: note: (near initialization for ‘_cffi_struct_unions[133].alignment’)
    error: command 'x86_64-linux-gnu-gcc' failed with exit status 1
    
    ----------------------------------------
  Can't rollback cryptography, nothing uninstalled.
Command "/usr/bin/python -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-
BiNMbi/cryptography/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" install --record /tmp/pip-d3EdLu-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /tmp/pip-build-BiNMbi/cryptography/

This is problem on Vbox 4.9.0-kali3-amd64

rabbauke, 2017-07-29:
try this

> I got the same error and fixed it this way.
> 
> sudo apt-get install -y python-pip
> pip install service_identity
> pip install scapy_http
> sudo apt-get install mitmproxy
> sudo ./installer.sh --install

alexisaeff, 2017-07-30:
Thank you! It's working!

closed 2017-07-30 by alexisaeff