P0cL4bs/WiFi-Pumpkin-deprecated #282: Error : WiFi-Pumpkin need PyQt4 :(

Wilanne, 2017-09-23:
Hi, I have a little problem with WiFi-Pumpkin, I did the installation but when I launch WiFi-Pumpkin I got this error : WiFi-Pumpkin need PyQt4 :(
I've the latest version of python3-qt4.. so I don't understand why I got this error.
And when I did the installation I got this : 
Collecting scapy (from -r requirements.txt (line 2))
  Using cached scapy-2.3.3.tgz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "/tmp/pip-build-pih1tggm/scapy/setup.py", line 36
        os.chmod(fname, 0755)
                           ^
    SyntaxError: invalid token
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /tmp/pip-build-pih1tggm/scapy/
----------------------------------------
[=] checking dependencies  
----[✔]----[+] hostapd Installed
----------------------------------------
[+] Distribution Name: Kali
----------------------------------------
[=]  Install WiFi-Pumpkin 
[✔] binary::/usr/bin/
[✔] wifi-pumpkin installed with success
[✔] execute  sudo wifi-pumpkin in terminal
[+] P0cL4bs Team CopyRight 2015-2017
[+] Enjoy

I use the latest version of WiFi-Pumpkin
My Laptop : Aspire 7739G with Kali installed on my hard drive
Network : Qualcomm Atheros AR9485 Wireless Network with ath9k driver and an Alfa Awus036NHA with AR9271 Chipset and ath9k_htc driver.

I'm French, I'm sorry for my bad English
Thank you
Willane

adolfaka, 2017-10-01:
sudo apt-get install python-qt4

sudo apt-get install libpq-dev python-dev libxml2-dev libxslt1-dev libldap2-dev libsasl2-dev libffi-dev libtiff5-dev libjpeg62-turbo-dev zlib1g-dev libfreetype6-dev liblcms2-dev libwebp-dev tcl8.6-dev tk8.6-dev python-tk

cd WiFi-Pumpkin

sudo pip install -r requirements.txt

sudo pip install service_identity

sudo pip install scapy_http

sudo apt-get install mitmproxy

mh4x0f, 2017-10-05:
> I've the latest version of python3-qt4.. so I don't understand why I got this error.

WP need python 2.7 version for work, install the 2.7 version and pip2 and follow the all command bellow.

closed 2017-10-06 by mh4x0f

jeretc, 2017-10-24:
error: WiFi-Pumpkin need PyQt4 :(

and error with building cryptography

i have python 2.7 installed.

[image]

shivgori, 2017-11-06:
I also face same problem python 2.7 already store in kali but WP shows pyqt4 error

rootsina, 2017-11-13:
have problem 
after install , :~# wifi-pumpkin
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 35, in <module>
    from core.loaders.checker.depedences import check_dep_pumpkin
  File "/usr/share/WiFi-Pumpkin/core/loaders/checker/depedences.py", line 2, in <module>
    from core.utility.collection import SettingsINI
  File "/usr/share/WiFi-Pumpkin/core/utility/collection.py", line 2, in <module>
    from PyQt4.QtCore import QSettings
ImportError: No module named PyQt4.QtCore

sabyasachiroot, 2017-11-17:
problem-No module named PyQt4.QtCore

mh4x0f, 2017-11-28:
@Wilanne I added a commit for solve this, but probably the next error will be some module that is not installed. you can find on google or issue WP  the solution.

Brain2000, 2017-12-06:
I'm getting the same issue in Kali-Linux 2017.1.  I manually modified the x509.h file so openssl wouldn't get the errors, but it seems to still fail with exit status 1 even though the only thing I see are deprecated warnings (similar output as #304 )

When trying to run it, the "WiFi-Pumpkin need PyQt4" message appears.

Brain2000, 2017-12-06:
It looks like if I change the line from:

  from core.utility.application import QtGui

to:

   from PyQt4 import QtGui

then it works.  ApplicationLoop is not importing from either line.

ethiccode, 2017-12-07:
Any help i either cant solve it.
it keeps on saying WiFi-Pumpkin need PyQt4
where should i change the code?
[image]
[image]

ethiccode, 2017-12-08:
sorry,i have reinstall but the error is same.
WiFi-Pumpkin need PyQt4 :(
@mh4x0f

yudevan, 2017-12-08:
Based on my experience, the best way to debug requirement for PyQt is to take out the import statement from the try except block. so you will know exactly what was causing the problem.

inside wifi-pumpkin.py change the lines below 
`try:
        from core.utility.application import ApplicationLoop
    except ImportError:
exit('WiFi-Pumpkin need PyQt4 :(')`

to 
`
 from core.utility.application import ApplicationLoop
`

And you will see what causing the error

Aziz

ethiccode, 2017-12-08:
i tried it still same error

yudevan, 2017-12-08:
You should get at least some debugging messages why it cannot find PyQt4. 
I think @mh4x0f  has fixed the code based on @Brain2000  comment above. 
do another pull to update your tree, and it should be set

vickyabi, 2018-11-24:
please help me,
it displaying like this 
root@kali:~# sudo wifi-pumpkin
Traceback (most recent call last):
  File "wifi-pumpkin.py", line 29, in <module>
    import core.utility.constants as C
  File "/usr/share/WiFi-Pumpkin/core/utility/constants.py", line 72, in <module>
    TEMP_CUSTOM = dir_path+'/templates/phishing/custom'
NameError: name 'dir_path' is not defined
root@kali:~#

mother2110, 2018-12-17:
I have tried everything on here to fix this issue. Any clear directions on what to do?

[root:~/pentesting/wifi/WiFi-Pumpkin]# wifi-pumpkin                                                                         (master) 
You do not have a working installation of the service_identity module: 'cannot import name opentype'.  Please install it from <https://pypi.python.org/pypi/service_identity> and make sure all of its dependencies are satisfied.  Without the service_identity module, Twisted can perform only rudimentary TLS client hostname verification.  Many valid certificate/hostname mappings may be rejected.
WiFi-Pumpkin need PyQt4 :(

I have installed service_identity module and PyQt4 is installed.

hackvice, 2019-12-11:
Have an error [!] WiFi-Pumpkin need Python 2 :( " But im having 
python -V
Python 3.7.5
Help?